sinara-hw/Booster #180: Simplify firmware

hartytp, 2019-03-19:
@wizath firstly, a massive thanks for all the work you've done on the firmware! We're getting there.

I had a look at the code and it seems it's grown a little complex. At least some of that is due to having to support multiple hardware versions because the software has had to work around bugs we've found in the hardware.

#177 is the last remaining "critical" issue that's preventing us from using Booster. I suggest that once that issue is resolved, we should pause for a second and do some housekeeping on the firmware.

The good news is that with the latest fixes, I don't think the hardware has any more "features" that cause issues. So, I'd like to suggest that we completely drop support in the firmware for all versions of the hardware <1.4 (i.e 1.3 after applying the known fixes). That should make things easier.

One reason I'm happy to drop support for old versions is that the FET death issue means that the older versions are going extinct quite rapidly. Soon there will be none left to support.

In practice, we'll sort out having our older versions patched/replaced with newer versions.

What do you think?

hartytp, 2019-04-09:
@wizath when you're happy that all no-longer-needed features are removed (rather than just commented out) and the fimrware is cleaned up, please close this issue.

hartytp, 2019-06-03:
@wizath is this done now? Can we close?

wizath, 2019-06-03:
Yes, unless I overlooked something, all of the legacy code is now removed.

closed 2019-06-03 by wizath

hartytp, 2019-06-03:
Cool, thanks! Getting there now :)