sinara-hw/Booster #181: v1.4 rc1

hartytp, 2019-03-19:
@gkasprow We've now finished testing Booster v1.3 Other than the outstanding software issues, it seems to work really excellently. A massive thanks to everyone who worked on it.

While it's fresh in our minds, can we sort out a v1.4 rc of the project with the known fixes applied.

Once that's done I'll start discussing with TechnoSystem the possibility of upgrading and repairing our older Boosters.

gkasprow, 2019-05-26:
v1.4 changelog:
-  IC20 on mainboard marked as DNP (#187)
- added LDO foldback info to the schematics (#176)
- increased opamp output resistors (R17,R41,R27) from 22R to 100R (#175)
- added note and increased capacitances (C27,C26,C26) from 1nF to 10nF(#170)
- replaced DFF with faster one (#168)
- changed C52 from 100nF to 1uF (#134)
- reset signals for detectors shorted to GND (#166)
- fixed production issues (#148)
- TYCO_5750489-1 is obsolete, HARTING_09561617712 was proposed
- added 3.3nF sampling caps (#143). Footprint was already there.
- grounded RST in of the power detector (#167 )
- added hysteresis RC 0 100p, 1k, 0R (#164 )
- added 10u cap at the LDO output (#125 )

gkasprow, 2019-05-27:
v1.4 release is here https://github.com/sinara-hw/Booster/releases/tag/v1.4rc1

closed 2019-05-27 by gkasprow

hartytp, 2019-05-27:
Thanks Greg! Did TechnoSystem + Creotech + @wizath confirm that this fixes all issues they're aware of?

hartytp, 2019-05-27:
@gkasprow one cosmetic thing:

on the "main board" schematic, on the RF module pages, there are a lot of signals that snake across the diagram. It would be much clearer if those wires were broken and just had net labels at each end.

Also, why isn't the output overdrive label near C83??

At this stage, it would probably be cleanest to completely remove IC13 and all components related to it, rather than DNPing them. These will never be used again, since the OpAmp scheme works much better!

hartytp, 2019-05-27:
As an example of a long net that would be easier to follow if broken and labelled on each end: the trace going to the gate of T5. Just break it and label each end something like "BIAS_EN"

hartytp, 2019-05-27:
Another example is the SDA and SCL lines. It's kind of hard to follow them across the entire schematic page.

gkasprow, 2019-05-27:
I was thinking about making general cleaning in this project. The RF modules also become messy and they need to be redrawn with high-quality CERN libraries. But this takes time which I recently don't have too much due to other activities. So this has to wait.

hartytp, 2019-05-27:
@gkasprow that's fine. These are all cosmetic issue so we can leave them for a later date. The project is actually in quite good shape at the moment and these are all minor nits!

hartytp, 2019-05-27:
@gkasprow currently the bias set DAC uses P3V3 as its reference. That line powers quite a few other things (even the micro) are we happy using it for such a critical reference?

hartytp, 2019-05-27:
It's also just the unfiltered output of a SMPS, so will be really quite noisy!

hartytp, 2019-05-27:
Also, not that it's worth changing now, but why didn't we add ferrite beads to all lines entering/exiting the RF modules (e.g. I2C, TTLs)? Wouldn't that help to keep the high-frequency cross-talk down?

hartytp, 2019-05-27:
(they cost almost nothing to add).

hartytp, 2019-05-27:
@gkasprow a general engineering question for you: IC21 states that it requires a minimum output capacitor of 10uF for stability. You've used 10uF ceramic, which will probably be a couple of uF once the DC bias is applied. Does this mean that it won't be stable, or does the 10uF number in the data sheet include plenty of slack to accommodate decrease in capacitance with voltage/temperature (ie. does the data sheet really mean that it needs 10uF of capacitance, or does it mean use a capacitor that says 10uF on the package?)

hartytp, 2019-05-27:
@gkasprow out of curiosity, did someone from creotech ever do a design review of Booster? e.g. checking basic things like absolute max rating of all IC pins, required capacitors, etc? I'm not aware of any issues and everything I've checked in the new release looks spot-on, but just curious how many eyes have been over the design at this stage.

hartytp, 2019-05-27:
@wizath NB you may have to tweak the ADC sampling time after changing the RCs.

gkasprow, 2019-05-27:
There was no such review done in Creotech yet. People are still saturated with Sayma boards.
Guys were examining RF modules so far.

There are ferrite beads on all signal lines entering the module, on the module side.
There are also capacitors or footprints on the main board's side.
IC21 has 16V 10uF capacitor working at 5V, so the value did not drop that much. There are also other caps on the RF modules connected by 0.15R series resistance which is more or less what tantalum capacitor would do. So, I wouldn't worry about stability.  But I can increase it to 22uF.

hartytp, 2019-05-27:
@gkasprow the DFF is only specified down to 5V. I think that's fine as it should still toggle in <1us, but might have been nice to use one that was specified at 3V3

gkasprow, 2019-05-27:
You are right. We should use at least FB for IC15 DAC supply. The 3V3 rail for all modules should be generated by LDO since there are instrumentation amplifiers and ADCs.

gkasprow, 2019-05-27:
DFF works from 3V, but its performance is not known.

hartytp, 2019-05-27:
> There are ferrite beads on all signal lines entering the module, on the module side.
There are also capacitors or footprints on the main board's side.

Aah yes, okay, that's fine :)

> There was no such review done in Creotech yet. People are still saturated with Sayma boards.
Guys were examining RF modules so far.

Understood. Well, it's pretty well tested now so I expect that we have caught most of the major issues. However, if Ceotech/Technosystem want to sell this as a product (and I've talked to a lot of groups who are interested in it) then it might be worth their while doing a full design review first. It's quite a complex design so plenty of places for small issues to hide and cause problems.

hartytp, 2019-05-27:
> DFF works from 3V, but its performance is not known.

Yes, it will work, question is just how slow. If it is very slow then the sequencing can get complex (which was part of the issue with v1.3). I think it will be fine in this case, but it's hard to verify the reset sequencing if the switching times are not specified at our operating voltages.

gkasprow, 2019-05-27:
True, but take into account that the commercialization effort is really huge. Look at how many boards we had built during the last 2 days. It's nearly 30! And careful design review takes a week or so.  We will certainly do it.

In the case of DFF we can replace it with HC-series, but different footprint.

hartytp, 2019-05-27:
> True, but take into account that the commercialization effort is really huge. Look at how many boards we had built during the last 2 days. It's nearly 30! And careful design review takes a week or so. We will certainly do it.

Yes. This is why many commercial products have small bugs! (Even Agilent synths tend to have their share of bodge-wires inside them). I'm not demanding it since the boards work, but it's something to think about for the future (design reviews are cheaper than support!).

hartytp, 2019-05-27:
The best would be if we/Creotech/whoever can get a grant to fund design reviews of the complete Sinara family, writing docs, etc.

hartytp, 2019-05-27:
@gkasprow I've finished my design review and posted on all the issues I've seen. Other than that, it all looks good! Thanks again.

hartytp, 2019-05-28:
double checked a few things, but now I'm done with the schematics.