sinara-hw/Booster #182: Output interlock calibration does not persist over reset

cjbe, 2019-03-20:
I calibrate the interlocks manually, and set the parameters using the `intparams` VCP command. The interlock power calibration is then correct. I restart the Booster, then the interlock calibration is lost. However, the correct calibration is still shown in the VCP status table:

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 0.00 CURRENT: 0.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            B2:13   B1:CF   90:A2   69:E0   6D:76   56:81   3C:9C   39:6B
TXPWR [V]       0.01    0.03    0.01    0.01    0.01    0.01    0.02    0.01
RFLPWR [V]      0.00    0.01    0.01    0.01    0.01    0.01    0.02    0.02
TXPWR [dB]      -7.30   -7.36   -6.31   -7.01   -7.64   -7.48   -7.80   -6.80
RFLPWR [dB]     -2.80   -2.13   -3.56   -4.33   -4.47   -5.15   -5.86   -5.81
I30V [A]        0.095   0.145   0.142   0.170   0.002   0.125   0.105   0.132
I6V0 [A]        0.151   0.090   0.160   0.191   0.092   0.083   0.003   0.074
5V0MP [V]       4.985   4.976   4.966   4.971   4.959   4.962   4.951   4.973
ON              0       0       0       0       0       0       0       0
SON             0       0       0       0       0       0       0       0
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            10      54      9       9       14      18      29      15
ADC2            8       8       8       10      9       13      38      37
DAC1            4095    4095    4095    4095    4095    4095    4095    4095
DAC2            0       0       0       0       0       0       0       0
SCALE1          83      87      87      85      85      86      86      87
OFFSET1         615     695     558     605     658     661     681     599
BIASCAL         1811    1960    1690    1700    1712    1687    1657    1756
HWIS            0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
HWIO            0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00
LTEMP           24.00   24.00   26.50   24.00   25.00   24.00   25.00   25.25
RTEMP           25.25   25.25   24.00   25.00   26.25   25.25   23.00   23.00
==============================================================================
```
However the interlock calibration does not appear to be loaded correctly by the "set interlock" command:
```
> intv 0 30
[intv] Interlock value for 30.00 = 0
```

After resetting the interlock parameters, the "set interlock" command works again:
```
> intparams 4 85 658
> intv 4 30
[intv] Interlock value for 30.00 = 3208
```

hartytp, 2019-03-21:
@wizath any luck looking into this? This and #183 currently make our v1.3 Booster unusable which is causing us lots of problems (our older versions of Booster are dropping like flies as the FETs die).

wizath, 2019-03-21:
Oh, overlooked that issue, will fix in a minute

wizath, 2019-03-21:
Fitting 8 byte double into 4 byte wasn't exactly wise choice. Fixed.
https://github.com/sinara-hw/Booster/issues/98#issuecomment-475208961

cjbe, 2019-03-21:
This works - thanks!

closed 2019-03-21 by cjbe