sinara-hw/Booster #184: firmware: scrap software OVC

hartytp, 2019-03-21:
@wizath now that we have hardware foldback current limiting, I do not think that the software OVC makes sense. The threshold one needs to set to allow the channels to operate reliably at max power is sufficiently close to the hw limit anyway.

How about this: we trust the foldback limit. The firmware just detects if the channel is on but the current is too low (indicating a serious error like a dead FET or foldback limit tripped). What do you think?

hartytp, 2019-03-21:
Actually @wizath I'm not even sure that works either. Once the foldback trips, the supply voltage drops. If it drops to less than 3V3 then the InAmp error becomes huge. So, I'm not sure that I see *any* means of making a reliable interlock without changing the hardware.

@gkasprow can you think of any way around this?

If not I think I'm okay removing the OVC completely. The new hw current limits seem to work really well and I haven't seen anything apart from false positives from the software OVC. Simplifying the firmware is always good. What do you think @gkasprow @wizath ?

wizath, 2019-03-21:
If you trust hardware foldback, we could disable it. 
IMO SW OVC uses some dirty tricks in order to work, so removing it would be a plus.

cjbe, 2019-03-21:
Let's kill it.

cjbe, 2019-03-21:
@wizath PS if there are any other features that require dirty tricks to get working then please let us know, as we might be able to scrap them and simplify things.

cjbe, 2019-03-21:
@wizath anyway, once this is done, please close this issue.

closed 2019-03-22 by wizath