sinara-hw/Booster #186: Booster occasionally fails to respond to SCPI queries

cjbe, 2019-03-22:
Occasionally the Booster fails to respond to SCPI queries. When this occurs there are no errors in logstash, and the interface operates normally for subsequent connections.

In this [packet dump](https://github.com/sinara-hw/Booster/files/2996607/dump.zip) the Booster responds to a series of queries successfully, then after the `MEAS:REV? 4` query in packet 746 fails to respond (just sends a zero-length ACK).

wizath, 2019-03-25:
Modified IRQ handling method a bit and fixed that locking bug, so it should behave nicely now.

cjbe, 2019-03-25:
With the updated firmware in https://github.com/sinara-hw/Booster/issues/98#issuecomment-476200623 I still see this problem. Using the dumb script below the Booster typical stops responding after ~100 queries.
(However it only takes ~1ms to respond to the queries, which is awesome!)

```python
import socket
import time

addr = "10.255.6.233"
sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.connect((addr, 5000))
sock.setsockopt(socket.SOL_TCP, socket.TCP_NODELAY, 1)

def query():
    sock.send("CHAN:ENAB? 0\n".encode())
    with sock.makefile() as stream:
        response = stream.readline().lower().strip()
    return response

i = 0
while True:
    t0 = time.time()
    query()
    dt = time.time()-t0
    print("{}: {:2.1f} ms".format(i, dt*1e3))
    i+=1
```

wizath, 2019-03-25:
OK, I see where is the problem

wizath, 2019-03-25:
That sounds like some kind of race condition etc. For a while use something like 10 ms delay between calls and it should work fine. I'll try to catch this since it seems that Ethernet chip is having hard time managing so many calls.

hartytp, 2019-04-02:
Did you manage to identify the cause of this?

hartytp, 2019-05-29:
Nice.

That reminds me, can you push teh source to the main branch when you get a mo?