sinara-hw/Booster #187: Two EEPROMs mounted on mainboard

wizath, 2019-03-29:
Also, they have different addressing methods while having same address so it crashes whole network configuration settings.

hartytp, 2019-04-13:
@gkasprow @wizath can this issue be fixed by just removing one EEPROM/changing a resistor to adjust the address of one EEPROM? If so, let me know what changes are needed and we'll apply them to the boards we have. Also, can you ask Creotech to make these changes on the rest of the units they're producing for us please?

hartytp, 2019-04-13:
To do: once this is fixed in hw, verify that this issue is gone https://github.com/sinara-hw/Booster/issues/156

hartytp, 2019-04-15:
ping: did someone alert Creotech about this?

kaolpr, 2019-04-15:
I see that the only solution is to remove one of the EEPROMs (24AA02E48 has don't care at 3 least significant bits of the address). Which one should be not mounted?
@wizath @hartytp

wizath, 2019-04-15:
Remove the one without UUID

kaolpr, 2019-04-15:
Ok, so IC20 not mounted. 
@hartytp is that ok for you? As those units are already assembled we need to unsolder those chips.
@gkasprow Could you update design documentation?

hartytp, 2019-04-15:
That’s fine for me so long as it’s okay with @wizath.

closed 2019-05-26 by gkasprow