sinara-hw/Booster #190: wiki: interlock description

hartytp, 2019-05-08:
@wizath is the description of the interlock on the wiki accurate? Are all interlocks described on the wiki? Do all interlocks function *exactly* as described on the wiki? If so, please close this issue.

wizath, 2019-06-13:
I've added more detailed information about interlock levels, you can adjust them it if you like

hartytp, 2019-06-13:
Thanks, I'm happy so long as
- (a) all interlocks on booster are listed on the Wiki
- (b) there is no incorrect info about the interlocks on the Wiki

If that's done, please close. Thanks!

wizath, 2019-06-14:
Fixed little mistakes, removed information about software OVC protection (since we decided to scrap it). Description is fine now I suppose.

closed 2019-06-14 by wizath