sinara-hw/Booster #191: pcb clean up

hartytp, 2019-05-27:
- [ ] annotation about InAmp resistors is no longer correct. Can you also add a on-line heading to this note so that it's clear what this is about (e.g.'Current sense resistors:'). Also, is it worth adding a note that the InAmps cannot be trusted when the outputs are not correctly biassed (or just say that care must be taken to operate in correct regime, see data sheet)

hartytp, 2019-05-27:
- [ ] redraw RF modules using cern lib (https://github.com/sinara-hw/Booster/issues/181#issuecomment-496298222)

hartytp, 2019-05-27:
- [ ] on the "main board" schematic, on the RF module pages, there are a lot of signals that snake across the diagram. It would be much clearer if those wires were broken and just had net labels at each end. e.g. the SDA and SCL lines should just be short wires with net names and not long traces across the schematic page. Similarly for the gate of T5 (should just be a labelled net something like 'BIAS_EN')

hartytp, 2019-05-27:
- [ ] why isn't the output overdrive label near C83??

hartytp, 2019-05-27:
- [ ] At this stage, it would probably be cleanest to completely remove IC13 and all components related to it, rather than DNPing them. These will never be used again, since the OpAmp scheme works much better!

hartytp, 2019-05-27:
- [ ] remove power detector internal comparator components (inc IC1A). We're never going to use these again and they just add confusion to the schematic. We can also remove the IN_RST etc lines from the headers.

hartytp, 2019-05-27:
- [ ] In an ideal world, it probably also makes sense to move the bias voltage generation to inside the supplemental PCBs. There are a few reasons for this, like supporting different RF module bias requirements in the future without changing the main PCB, making it easier to fix issues in the bias voltage generation (just change the supplemental control PCB), not needing as many signals going into the RF modules (just use existing I2C bus). Not a high priority, just something we could do one day during a tidy up

hartytp, 2019-05-27:
- [ ] on the supplemental control PCB the out_bias is labelled 10mA. I doubt the OpAmp can support that and it's not consistent with the annotations on the main PCB. I think there are now a few inconsistencies in these annotations, which we should fix one day

hartytp, 2019-05-27:
- [ ] why is the input power measured on a I2C ADC and the output forward/reverse power measured on a different ADC? Wouldn't it be simpler to just put a multi-channel I2C adc on the supplemental control voltage and measure everything on that. Again, reduces the number of wires entering/leaving the module

hartytp, 2019-05-27:
(NB these are not things to fix now, but worth considering if we ever make major changes to the design)

hartytp, 2019-05-27:
- [ ] put crosses through DNPd components on RF board schematics
- [ ] strip out all components associated with the power detector's internal comparators

hartytp, 2019-05-27:
- [ ] remove bias pot instead of DNPing

gkasprow, 2019-05-29:
> In an ideal world, it probably also makes sense to move the bias voltage generation to inside the supplemental PCBs. There are a few reasons for this, like supporting different RF module bias [...]

That's true. But we don't have that much space on the board. Tomasz was afraid of putting many components there and wanted to leave the solid copper plane above the RF section. That's why components are "crowded" close to the connectors. We can always increase the regulation range easily by changing the resistors. We would also need to pass the power sequencing signal to the RF module.

> why is the input power measured on a I2C ADC and the output forward/reverse power measured on a different ADC? Wouldn't it be simpler to just put a multi-channel I2C adc on the supplemental control [...]

We wanted to have a quick reaction of the CPU so internal ADC was used. External ADC over I2C is very slow. At this moment it requires a lot of work in software to change it and we get nothing in reward since the CPU ADCs are already there.

Every design can be improved, but with every improvement, there are new issues.
I propose only cleaning the schematics, use high-quality libraries.
Once we decide to use other RF modules, we will do the necessary changes.

gkasprow, 2019-05-29:
- [ ] we should use a continuous namespace for all designators to make sure that there are no multiple C1 capacitors in Booster project.

hartytp, 2019-05-30:
Makes sense.

I would still like to see the dnpd componets which will never be used removed. Nice and quick and saves some board space.

Tidying up the nets on the schematic is nice but not worth it if it will take too long.

hartytp, 2019-06-13:
breaking out into separate issues since most of this is now done

closed 2019-06-13 by hartytp