sinara-hw/Booster #192: comparator hysteresis

hartytp, 2019-05-27:
- [ ] the way we implemented it, the capacitor and resistor were in series, not parallel. That way, there was no DC current draw, which could affect the measurement made by the ADCs...

gkasprow, 2019-05-27:
Hm. That makes sense. But I remember the discussion some time ago when somebody convinced me that it should be in parallel. That's why I was afraid of CPU ADC measurement and was so worried about threshold levels precision. And that's why I proposed to increase the values.

hartytp, 2019-05-27:
Interesting, I’d love to know what the concern was. We put them in parallel in the boards we fixed and they work fine.

gkasprow, 2019-05-28:
so shall I place them in series or in parallel?
If we place them in series, the hysteresis will be only for AC signal.
If we place them in parallel, the hysteresis will be very strong, for both AC and DC but will influence CPU ADC measurements.

hartytp, 2019-05-28:
@gkasprow I vote for series and here is why:
- the reason the hysteresis is required is that the DFF is slow compared with the comparator, so it's possible for the comparator to oscillate for long periods of time without the DFF ever triggering properly
- with AC hysteresis, the frequency of the oscillations is lowered (set by the RC time constant). So long as the frequency is low compared with the max toggle rate of the DFF (which should ideally be specified at our working voltage), the DFF will suppress the oscillations.

hartytp, 2019-05-28:
@gkasprow NB the current circuit does not quite fix all the problems. I believe that there is still the possibility of a race when the DFF is reset. The problem occurs partially because the DFF reset is global and applies to all channels at the same time. So, it is possible for the interlock to trip while the DFF reset line is pulled high by the microcontroller (in which case, the DFF will not see the edge of the comparator and will not latch).

This is likely to be a very rare case, but still worth considering.

hartytp, 2019-05-28:
Would this be fixed if we tie the DFF S and R lines together? That way whenever the DFF is reset the RF switch is opened, ensuring that the comparator output goes low. Since the comparator (after hysteresis) and power detector are slow compared with the DFF reset time, this will ensure that the interlock can only retrigger once the DFF is ready. What do you think?

gkasprow, 2019-05-28:
with S and R lines tied together the DFF output could be undetermined.

hartytp, 2019-05-28:
> with S and R lines tied together the DFF output could be undetermined.

At least for the CD4013BC, the truth table specifies the output when S=R=1 (Q=Q_bar=1). There could of course be a transient depending on which pin goes high first

gkasprow, 2019-05-28:
Yes, but after they are released the state could be unknown due to the race condition.

hartytp, 2019-05-28:
True. I guess that suggestion is also nasty since it means that all outputs glitch when the interlock is reset.

The other idea I had was to put an RC filter (something like 1k + 300pF) between the comparator output and the DFF input. Then add a FET on the DFF input that pulls the DFF input low when the RESET line is high. The RC filter doesn't have a significant impact on the interlock speed, but ensures that whenever the DFF is reset, its input is LOW so we never miss a transition.

gkasprow, 2019-05-28:
let's put fast DFF from LVC74 series. it has a few ns of propagation delay.

hartytp, 2019-05-28:
Sounds good. What about a FET with small RC filter to ensure correct sequencing on RESET?

gkasprow, 2019-05-28:
DFF needs short rising edge. If you use LPF then it will toggle the output when clock is high but then toggle it back when clock is low. I had such issue in the past

hartytp, 2019-05-28:
true. Okay, so then...next try: 1k resistor between comparator and DFF, no capacitor. FET to pull the DFF input low during reset and some kind of delay on the FET gate to ensure it switches after reset (even an inverter should add enough delay for this).

hartytp, 2019-05-28:
Or, @gkasprow suggest a suolution that you think will work

hartytp, 2019-05-28:
@gkasprow how about we fix this in SW? We can do the following: we make it part of the contract that during interlock RESETS all output channels switch off briefly. So, the interlock reset sequence is:
- all RF switches open
- delay of at least a few us. we now know that all DFF inputs are LOW
- RF switch override releases, so switches close again
- any interlock conditions will now guarantee to trigger robustly

any objectsions @cjbe @wizath?

wizath, 2019-06-03:
> during interlock RESETS all output channels switch off briefly

Correct me if I'm wrong but don't we currently use only one line for RF switch & interlock reset? ATM we have procedure like
- set proper bias voltage (with sequencing)
- disable RF switch (even if it's disabled by interlock circuit)
- wait a little bit
- enable RF switch again

hartytp, 2019-06-03:
@wizath okay, so when one presses the RESET button/issues a software reset, the switches are all opened (no RF). Is that correct?

If so, I think we're okay and there can be no race condition on the output interlock, so long as the interlock is held high for at least a few us whenever it is reset since:
- To reset the interlock the switches must open, disabling the RF
- So long as the switch is open for at least a few us, the RF power meter has a chance to respond and the DFF input will also be LOW when the RESET is released
- due to the filter/hysteresis components we've added, the DFF is guaranteed to come out of RESET before the comparator output goes HIGH, so the next rising edge will always be captured

For the input interlock I guess there can still be a race, since the power detector is before the switch. Not much we can do about that. But it shouldn't be too much of a problem since the output interlock will still trip.

So long as you're happy that this is all fine and there is no possibility of a race, we can close #200

hartytp, 2019-06-12:
@gkasprow did you implement this?

gkasprow, 2019-06-12:
Let's use the [NC7SZ175P6X](https://www.onsemi.com/pub/Collateral/NC7SZ175-D.PDF) DFF. It has only reset signal which is inverted but we can use the signal before the IC2 inverter

gkasprow, 2019-06-12:
[](url)
[image]

hartytp, 2019-06-13:
Can we make that 100p and 1k, please? Otherwise it's quite a lot of current for things to source.

gkasprow, 2019-06-13:
done

closed 2019-06-13 by gkasprow