sinara-hw/Booster #193: bias DAC reference

hartytp, 2019-05-27:
should be a 3V3 LDO or even a buffered reference. We could also just move this DAC + OpAmp to the supplemental control PCB where there is already a LDO.

> You are right. We should use at least FB for IC15 DAC supply. The 3V3 rail for all modules should be generated by LDO since there are instrumentation amplifiers and ADCs.

gkasprow, 2019-05-27:
It's too much work. One LDO per mainboard won't cost us much and we keep compatibility with previous versions.

hartytp, 2019-05-27:
Sounds good. Let’s make sure that nothing with high current draw runs from it so we don’t get high voltage drops

hartytp, 2019-05-27:
Hmm thinking about this more I really think we should run the dac off a reference rather than an LDO (or at least an LDO like the 3042 which has reference quality performance down to a Hz or so). The bias is the single most important aspect of the design so we should put some care into it

hartytp, 2019-05-27:
Also let’s make sure that nothing draws current spikes from that rail which could end up on the rf

gkasprow, 2019-06-12:
I added 3042 LDO and supplied  DACs. ADCs are already supplied from 2V5 reference.

closed 2019-06-12 by gkasprow