sinara-hw/Booster #197: InAmp resistors

hartytp, 2019-05-28:
@gkasprow didn't we increase the InAmp resistor values at some point? The InAmp data sheet suggests that Vsense minimum should be 20mV. IIRC the bias current is currently 50mA, and the current 28V sense resistor is 90mR on the scehamtic, so voltage is 4.5mV which won't work.

hartytp, 2019-05-28:
same goes for the 5V. What's the current draw and doesn't that need a larger sense resistor to give >20mV sense voltage at the InAmp?

gkasprow, 2019-05-28:
the power stage current monitor sense resistor was previously 20mR and was increased to 90mR.
It was optimal for previously used range, so it's a good idea to increase it.
@wizath what part of range do we use for 5V rail current measurement?

hartytp, 2019-05-28:
Cool. Let’s increase it then. The fw will have to look up the calibration based on the hw rev

hartytp, 2019-05-28:
Just make sure the sense resistor doesn’t die at max current.

hartytp, 2019-05-29:
@gkasprow I don't like making changes this late in the game, but I've thought about this a lot and I'm just not happy with the current sense InAmps we're using in this design. The architecture they use is not well suited to the parameters of our design, which causes all these strange issues.

One option would be to use https://www.analog.com/media/en/technical-documentation/data-sheets/6106fb.pdf The InAmps could all be run from the main 30V power supply.

Other options e.g. https://www.analog.com/en/parametricsearch/11082#/p2839=2.4|3.3&p4125=30|100&p4126=-24|5

What do you think @gkasprow?

hartytp, 2019-05-29:
This also allows us to keep the sense resistor relatively small so we don't increase the supply impedance which is nice...

gkasprow, 2019-06-12:
They have exactly the same footprint, so let's go for LT6106CS5#TRMPBF. I need to add the load resistor only.

closed 2019-06-12 by gkasprow

gkasprow, 2019-06-12:
what about 400mA range for both supply rails?

hartytp, 2019-06-12:
For the 28V the max should be a little higher than the LDO fold back limit .

What is the max on the other rail in the power budget?

gkasprow, 2019-06-12:
True. Let' set it to 600mA
The 5V rail consumes typically 220mA with no RF power.

gkasprow, 2019-06-12:
won't the 0.1R output impedance be too high for the PA? We didn't observe any issues related to that fact, did we?

hartytp, 2019-06-12:
What sense resistor do you suggest?

gkasprow, 2019-06-12:
I used 0.1R for both rails.

hartytp, 2019-06-12:
0.1 ohm gives V sense of 5mV at our normal 50mA current. In amp offset is 0.25uV so 5% if we make the sense resistor any smaller we either need to find a more accurate InAmp or include the sense resistor in the LDO feedback loop.

If you’re worried about the 0.1ohm supply rail we can do that, I’m not sure if it’s necessary or not.

In any case what’s the resistance of the FBs pcb traces, connectors etc on that rail. Must be quite some 20s of mOhm right so is this sense resistor so much larger that the other resistances?

gkasprow, 2019-06-12:
I'm not worried. There are ferrite beads that have similar resistance.

hartytp, 2019-06-12:
@gkasprow i guess the device performance is not that sensitive to bias current (and in any case the FETs have quite a large temp co). If we’re happy with 10% accuracy on this measurement then we can go for a 50mOhm sense resistor. 

As I said though we should consider the other impedances like FBs PCB traces, contacts etc since there is no point making the sense resistor smaller that the other resistances.

In any case, I trust your judgement on this so do what you think is best. Just add an annotation on the schematic saying what the InAmp sensitivity and accuracy are (assume typical Vos of around 150uV)

hartytp, 2019-06-12:
Okay, good! If you’re not worried then I won’t eother.

Do please make sure the schematic is annotated.

Let me know when the PDFs are ready and I’ll review