sinara-hw/Booster #199: signal ground FB

hartytp, 2019-05-28:
@gkasprow in the supplemental control board there should really be a ferrite on the signal ground pin to prevent RF leakage.

gkasprow, 2019-05-28:
do we observe excessive cross-talk? the FB introduces non-zero resistance, usually orde of tens of mOhm So it may break the bias setting accuracy causing positive feedback.

hartytp, 2019-05-28:
The cross-talk about 300MHz was getting a bit higher than we would have liked (fine, but an extra 10dB-20dB would have ensured it's not something I'll ever have to think about). 

I thought the SIGNAL_GND pin was only relevant for the power detector ADCs? AFAICT, all signal ground pins are shorted to the main board ground at the microprocessor and it has nothing to do with the bias setting.

gkasprow, 2019-05-28:
OH, you are talking about  SIGNAL_GND only, not the power supply. then OK, we can add FB, but anyway we have a ground loop here so this signal does not make much sense. It should actually be sampled by the CPU ADC for DC drop compensation :)

hartytp, 2019-05-28:
Yes. It seems like a nasty ground loop to me. For that signal to do good, wouldn't the CPU ADC need to sample all the SIGNAL_GND lines independently rather than tying them together?

Unless we really need it, I'd prefer to just scrap the signal completely. AFAIC currently it does no good and might do some harm.

gkasprow, 2019-05-28:
good idea, let's connect it to GND

gkasprow, 2019-06-11:
done

closed 2019-06-11 by gkasprow