sinara-hw/Booster #200: reset sequence

hartytp, 2019-05-29:
- [ ] in firmware, all outptus should be forced off when interlocks are reset to ensure that there can't be a race condition
- [ ] update docs (wiki) to note that this happens so that users aren't confused

wizath, 2019-06-03:
What do you mean by "all outputs should be forced off"?

hartytp, 2019-06-03:
@wizath maybe I've mis-remembered the HW configuration. I thought there was a CPU line that allows us to open the RF switches, so that the output power goes low.

wizath, 2019-06-03:
Yes, but that's the same line which will reset interlock when pulled down, so by resetting the interlock you automagically disable the RF switch for a second

hartytp, 2019-06-04:
@wizath if you're happy there is no possibility of a race during the interlock, please close this issue.

hartytp, 2019-06-05:
@wizath ping...

wizath, 2019-06-05:
Since when interlock occurs the RF switch is opened anyway, I think there is no need to write about internals of interlock reset sequence.

hartytp, 2019-06-09:
@wizath I’m pretty sure there is still a race with the input power interlock: since that’s before the switch the power does not go low during interlock resets so it will never retrip if it’s reset when the power is high. Can we just fix this in software? E.g. check the power before and after reset (to guard against the power going high *during* the reset) in sw and then interlock in sw as well as hw? What do you think? Anyway fix however you want but if we’re going to have this interlock then we should make sure it works!

closed 2019-06-25 by hartytp