sinara-hw/Booster #201: calibration documentation

jbqubit, 2019-06-04:
Working through calibration documentation I ran into some snags. I edited wiki to clarify a couple obvious things -- please review. 

### Module current

- What do the numbers mean that are output during the calibration? 

- Should output and input on SMA be open/shorted/50Ohm? Does it matter? 

- what are units of <desired current> mA or A?

I sometimes see "[biascal] error, current too high 52 > 50". Add explanation of how to proceed if this error happens.  

> After these steps, you should see following result:

I see  `[biascal] done, value = 1753, current = 50`.

### Output interlock calibration

- My booster amplifier output saturates at 33 dBm so can't complete second calibration step at 35 dBm. Is it OK to use 33 dBm with command `intcal <channel number> 2 33.0`? 

- Is the high power point supposed to be well into saturation of the amplifier? If so it would be clearer to state this. 

- is this output indicative of success? 

[ointcal] done, interlock value = 2754
[intcal] Calibration step = 1 completed = 2754
[intcal] done, got 1-point cal val 2754 pwr 23

>After these steps, you should see following result:

I see no message indicative of success or failure after completing the pair of calibration steps. 
 
### Output & Reverse power monitoring calibration

- My booster amplifier output saturates at 33 dBm so can't complete second calibration step at 35 dBm. Is it OK to use 33 dBm with command `intcal <channel number> 2 33.0`? 

### Input interlock calibration

Once interlock is done.... result is 

[iintcal] done interlock value = 95
[cal] Calibration step = 100 completed = 95
[iintcal] trying value 104, status 1
[iintcal] done interlock value = 104
[cal] Calibration step = 10 completed = 104
[iintcal] trying value 112, status 1
[iintcal] done interlock value = 112
[cal] done, value = 112

hartytp, 2019-06-04:
@jbqubit I didn't realise you had a booster! What hw and fw version + hw vendor are you using?

wizath, 2019-06-04:
## Module current
> Should output and input on SMA be open/shorted/50Ohm? Does it matter?

It does not matter since the RF switch is open

> what are units of mA or A?

Units are in mA

> [biascal] error, current too high 52 > 50

This is related to module current exceeding 2% setpoint margin because DAC step is set a little bit high. It'll be fixed soon, but for now, you can just redo your calibration with slightly higher current eg. `52 mA` and it should be fine

## Output interlock calibration

> My booster amplifier output saturates at 33 dBm so can't complete second calibration step at 35 dBm. Is it OK to use 33 dBm with command intcal <channel number> 2 33.0?

Yes, It'll be completely fine.

> is this output indicative of success?

That is a message after completing one step of two. After completing two calibrations you should see output like
```
[ointcal] done, interlock value = 994
[intcal] Calibration step = 1 completed = 994
[intcal] done, got 2-point cal val 994 pwr 36
[intcal] done, got power factors a=8.91 b=-25.49
```
## Output & Reverse power monitoring calibration

> My booster amplifier output saturates at 33 dBm so can't complete second calibration step at 35 dBm. Is it OK to use 33 dBm with command intcal <channel number> 2 33.0?

Yes, it's fine.

Regarding your previous email about RFL PWR being too high, there was an issue where reflected power was calculated based on wrong ADC measurements. https://github.com/sinara-hw/booster-firmware/commit/340d759e42bf00008eed65d01684ee634d7306a0 fixes that.

## Input interlock calibration
Could you try running input calibration on newest firmware? Since that should now be fixed. Attaching the .dfu file for easy upgrade

[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/3252859/eth-scpi.dfu.zip)

hartytp, 2019-06-04:
> My booster amplifier output saturates at 33 dBm

That's quite concerning isn't it?

hartytp, 2019-06-05:
@jbqubit can you have a go at updating the Wiki with any missing information and then close this issue, please?

hartytp, 2019-06-05:
Also, @wizath can you review the changes @jbqubit made to the wiki and fix as necessary, please? Thanks!

hartytp, 2019-06-05:
> RFPA v1.20
> HW v1.10

@jbqubit to confirm are you using a TechnoSystem (black box) Booster with revision <1.3?

jbqubit, 2019-06-05:
For what I reported above I was running `RFPA v1.20, built Apr  1 2019 16:13:27, for hardware revision: v1.10`

hartytp, 2019-06-05:
@wizath that version string is quite confusing. I assume `RFPA` is a software revision? Can you make that a bit clearer?

@jbqubit Can you double check the hw rev number engraved on the chassis, please? HW rev v1.1 is pretty old (we're preparing v1.4 atm) and has an issue that kills the FETs (might be why you only get 33dBm out at saturation). Unless the issues have been manually patched, those units will die and need to be upgraded to a more recent version.

jbqubit, 2019-06-05:
What's engraved on the chassis is "v1.3". The main board PCB says "v1.3". Why does firmware report v1.2? 

I'd do firmware upgrade but `dfu-util -l ` returns `dfu-util: Cannot open DFU device 0483:df11`.

hartytp, 2019-06-06:
@jbqubit if you have v1.3, was the interlock fixed? (e.g. add hysteresis capacitors to interlock comparator etc). Make sure those have been applied.

I'm also quite concerned that the max output power you're reporting is 33dBm, when it should be more like 37.5dBm...

hartytp, 2019-06-06:
@jbqubit what is the issue here? Can we close this?

closed 2019-06-06 by jbqubit