sinara-hw/Booster #203: version string

jbqubit, 2019-06-05:
Need clearer distinction between hardware and software version in `version` string. 

https://github.com/sinara-hw/Booster/issues/201#issuecomment-499251793

hartytp, 2019-06-06:
Nit: can you add a label to this issue, please?

wizath, 2019-06-06:
Already fixed in https://github.com/sinara-hw/booster-firmware/commit/11dfc1917682df3ffccc5d99b69498cc03ae7031

Scrapped hardware version from new software since it should be universal. New version includes commit hash and build date.
```RFPA v1.4.19157 98e655b, built Jun  6 2019 10:00:19```

Closing. Reopen if necessary

jbqubit, 2019-06-06:
Version string doesn't report on hardware version. Which is also relevant.

wizath, 2019-06-06:
https://github.com/sinara-hw/booster-firmware/commit/1a52edb38abc88deb494232939bdb0d3df6740b1

`RFPA v1.4.19157 2fe1161, built Jun  6 2019 16:29:13, hv rev 3`

Fixed.

closed 2019-06-06 by wizath