sinara-hw/Booster #204: output saturates at +33dBm

jbqubit, 2019-06-06:
Observed output saturating at +33 dBm for 160 MHz input. Hardware v 1.3. Software built Apr 1 2019 16:13:27. 

https://github.com/sinara-hw/Booster/issues/201#issuecomment-499362534

@hartytp said

>I'm also quite concerned that the max output power you're reporting is 33dBm, when it should be more like 37.5dBm...

Will check again once I figure out how to update firmware.

wizath, 2019-06-06:
> Will check again once I figure out how to update firmware.

Press DFU button, then power up the device to enter the bootloader mode
Alternatively, you can type `bootloader` in VCP to enter BL

Then use instructions listed in Wiki page to update the software.

jbqubit, 2019-06-11:
I updated firmware to v1.4.19155. 

Using a 20 dB attenuator designed to dissipate 10 W. R&S FSV40-N spectrum analyzer serving as power meter. Use 160 MHz RF source. Drive amplifiers into saturation. 
 
[image]

I opened this Issue with a measurement of CH0. After looking at all eight channels is looks like CH0 is an outlier.  

[image]

hartytp, 2019-06-11:
Are there some points missing in that plot around -2dBm for several channels?

What do the bias currents etc look like?

If I were you my next step I’m debugging this would be to manually measure the FET bias current by using a differential DVM to look at the voltage across the sense resistor on the main PCB. There is a known issue with the InAmps used for the bias current cal. If the bias is too low that could explain what you see.

hartytp, 2019-06-11:
I’d also ask TS if they saw this when they calibrated your unit.

@wizath any other ideas?

hartytp, 2019-06-11:
@jbqubit fwiw all the channels there look like there're saturating a bit early. c.f. some of the measurements Chris posted with our Booster v1.3. e.g. https://github.com/sinara-hw/Booster/issues/173#issuecomment-475312360 notice how the small signal gain should be something like 40dB +- 1dB, but at 35dBm output the gain should only have decreased by a little over 1dB.

hartytp, 2019-06-12:
@jbqubit a couple of observations about this:
1. I still wouldn't recommend that any other groups buy Booster until we've tested v1.4 (hopefully in a few weeks). There are still known issues in the current version that could explain things like this
2. While I rate TechnoSystem very highly as a supplier, we do need to think more carefully about QC for Booster than for the EEMs owing to the greater complexity. Creotech do full testing on each unit they produce and give a test sheet at the end, so they won't ship units with broken channels. We need to discuss with TS whether they have the capability of doing something similar. Otherwise, I'd recommend sticking with Creotech for Booster (TS are fine for all other Sinara designs AFAICT). @gkasprow do you have any comments about this

Anyway, back to the matter in hand. I think you have three options Joe:
1. Try to debug this. I'd start by measuring the power voltages and currents on a few channels using an independent DVM. If you find, for example, that the bias current is very low on channel 1 (due to InAmp offsets) then you can recalibrate for a higher nominal current until the actual current reaches 50mA (this will be fixed in v1.4).
2. Live with it. The power you get looks fine for most AOMs, and debugging is slow.
3. Send it back to TS and ask them to upgrade it to v1.4 electronics when they're ready.

Let me know which path you go for. But, for now I'm going to close this as I don't think there is anything to be done here and it's likely explained by one of the bugs we've already fixed (if this crops up again on a v1.4 unit then we'll debug properly). Feel free to re-open if you start debugging and find things you don't understand.

closed 2019-06-12 by hartytp