sinara-hw/Booster #206: RFLPWR > TXPWR

jbqubit, 2019-06-06:
Expectation is that RFLPWR < TXPWR. 

``` 
PGOOD: 1
FAN SPEED: 20 %
AVG TEMP: 27.19 CURRENT: 27.50
CHANNELS INFO
==============================================================================
#0#1#2#3#4#5#6#7
DETECTED11111111
HWID04:371E:53E3:ADC6:2DB4:80DF:033D:4306:1D
TXPWR [V]0.010.050.020.030.020.020.020.01
RFLPWR [V]0.080.010.110.040.140.030.010.01
TXPWR [dBm]-7.22-6.53-6.24-5.01-6.04-6.96-6.10-5.14
RFLPWR [dBm]-0.01-3.24-4.62-5.30-4.28-1.01-5.13-3.04
I30V [A]0.0520.0520.0540.0520.0520.0570.0520.054
I6V0 [A]0.2730.2550.2490.2510.2590.2530.2690.252
5V0MP [V]4.9915.0035.0094.9864.9894.9945.0005.005
ON11111111
SON11111111
IINT00000000
OINT00000000
SINT00000000
OVC00000000
ADC11173304531323110
ADC21281217367223461812
DAC121802180218021802180218021802180
DAC227543132325231403273317330963546
SCALE19068747375717084
OFFSET1662516492410483525457442
BIASCAL23791849175516171753168713911821
HWIS100.0069.0071.0070.0072.0069.0070.0082.00
HWIO454.00648.00696.00620.00681.00689.00576.00594.00
LTEMP25.2526.5026.5026.2526.0026.5026.0027.25
RTEMP25.2526.2525.0026.0026.0026.5027.0027.50
==============================================================================
p
> version
RFPA v1.4.19155 340d759, built Jun  4 2019 16:19:51

```

jbqubit, 2019-06-06:
Hardware is v1.3

wizath, 2019-06-06:
Reflected power is calibrated between 0 dBm and 35 dBm, so at low signal it could provide out of spec measurements.

hartytp, 2019-06-06:
Indeed. The power detector is only good down to ~-30dBm. According to the schematic +8dBm at the detector is with 38dBm output power. So, the power detector isn't expected to work below ~+8dBm output.

I see two issues here...

Firstly @wizath please can you update the documentation (and make sure Creotech/Technosystem are alerted) to state that calibration for both the forward and reverse power detectors should be done at +~36dBm and +~20dBm. Booster is a PA, so I'm not too fussed if the accuracy drops off below 100mW output and I wouldn't want to risk degrading the calibration accuracy at higher powers to increase the dynamic range.

Secondly @jbqubit maybe add a note on the wiki somewhere that power detector readings are only reliable down to ~+10dBm.

jbqubit, 2019-06-06:
I edited wiki. 

> So, the power detector isn't expected to work below ~+8dBm output.

TXPWR [dBm] and RFLPWR [dBm] could report "low" if < +10 dBm. Probably fine to keep reporting raw voltage on preceding lines.

hartytp, 2019-06-06:
> TXPWR [dBm] and RFLPWR [dBm] could report "low" if < +10 dBm. Probably fine to keep reporting raw voltage on preceding lines.

I'm not a fan of that approach. Having that kind of special case is annoying to handle e.g. if you're passing the logs you now have to make your code robust to dealing with numbers or strings.

@wizath what about clipping the power meter readings so they never read below some fixed threshold? Put the threshold as low as you can while still ensuring that the readings are reasonably accurate. Maybe 5dBm unless you think the detectors are still good to lower than that?

NB I obviously can't add up https://github.com/sinara-hw/Booster/issues/206#issuecomment-499531996 -30dBm at the power detector input corresponds to an output power of 0dBm. Although, in practice the readings might start becoming dodgy a little before then.

jbqubit, 2019-06-07:
How about something numerical and obviously nonphysical like -999 dBm? Somebody might be confused by seeing 5 dBm when it's really something much smaller.

hartytp, 2019-06-07:
@jbqubit No, I'd prefer to just clip to a sensible lower bound. Having the power suddenly jump to -999dBm is obnoxious. e.g. when one is plotting a log and the power suddenly changes by 1000 orders of magnitude, it tends to screw up the axes.

Clipping to a sensible lower bound is also, for example, what the Keysight V3500A does (they clip at -63dBm).

Beyond that, users have to recognise that there is no such thing in this context as "really much smaller" than 5dBm reflected in most practical cases. What's the directivity of the directional coupler in booster? How good is the output matching of Booster? Since the Booster power detectors can't know the output frequency, they only have a 2-point calibration, which can't correct for those kinds of effects (cf how involved the calibration for a real VNA is). In the world of RF, trying to look at 5mW of reflections from a 5W power amp (-60dB) is no mean feat. Users have to expect these kinds of things.

We can do some work to document this, but users have to have reasonable expectations informed by a little knowledge of RF (the Booster documentation cannot be an intro to RF).

hartytp, 2019-06-07:
Another way of putting it is that if you're really interested in <5mW reflections then Booster is probably not the right amp for you...

jbqubit, 2019-06-11:
OK. Clipping sounds good provided it's documented. 

Please close after making this change @wizath.

wizath, 2019-06-12:
Ok, but at what value would you like to clip the power measurements?

hartytp, 2019-06-12:
Whatever the lowest value where we think they’re still relatively accurate I’d guess somewhere between 0dBm and 10dBm but you’ve had more experience playing with these detectors

wizath, 2019-06-12:
I'll test them later this day and pick something

dnadlinger, 2019-06-12:
We'll still have the voltage readings unclipped, right?

wizath, 2019-07-15:
Since the detector linearity is quite good, power will be clipped below +5.0 dBm. At this point we still got like 0.8~1 dBm accuracy.

closed 2019-07-15 by wizath