sinara-hw/Booster #208: setting IP

jbqubit, 2019-06-06:
`ifconfig` displays info but doesn't seem to permit setting IP, network, mask. Is there a way to set these and save to EEPROM?

wizath, 2019-06-07:
To set static IP use following command
`ifconfig <ip src> <gw ip> <netmask>`
e.g.
`ifconfig 192.168.1.10 192.168.1.1 255.255.255.0`

It will write changes to EEPROM

hartytp, 2019-06-07:
Remind me how one switches back to DHCP? (Is this all covered in the help menu/wiki?).

wizath, 2019-06-07:
```
(2/2) help ifconfig
ifconfig usage:
        ifconfig - display network configuration info
        ifconfig dhcp - enable DHCP
        ifconfig <ip src> <gw ip> <netmask> - set static network address
```

hartytp, 2019-06-07:
Good, thanks. @jbqubit did you try looking at the help message?

closed 2019-06-07 by hartytp

jbqubit, 2019-06-07:
The help is not present in v1.4.19155. Glad to see that it's in the latest version.