sinara-hw/Booster #209: v1.4rc2 changelog

gkasprow, 2019-06-12:
v1.4 rc1 changelog:
    - IC20 on mainboard marked as DNP (#187)
    - added LDO foldback info to the schematics (#176)
    - increased opamp output resistors (R17,R41,R27) from 22R to 100R (#175)
    - added note and increased capacitances (C27,C26,C26) from 1nF to 10nF(#170)
    - replaced DFF with faster one (#168)
    - changed C52 from 100nF to 1uF (#134)
    - reset signals for detectors shorted to GND (#166)
    - fixed production issues (#148)
    - TYCO_5750489-1 is obsolete, HARTING_09561617712 was proposed
    - added 3.3nF sampling caps (#143). Footprint was already there.
    - grounded RST in of the power detector (#167 )
    - added hysteresis RC 0 100p, 1k, 0R (#164 )
    - added 10u cap at the LDO output (#125 )

v1.4 rc2 changelog:
- added 1uF cap to the CPU ref and VDDA
- removed components in PA_Channel that were DNP. i.e. negative LDO
- DACs and sensitive circuits are supplied from LT3042 LDO/reference
- cleaning the PA_Channel schematics, long wires replaced by nets
- on SUPL_CTRL board removed not used ADL reset circuit and logic gate

hartytp, 2019-06-12:
So the only remaing change is  the new InAmp I think

gkasprow, 2019-06-12:
I'm just doing that.

hartytp, 2019-06-12:
:)

hartytp, 2019-06-12:
Sorry thought you’d finished

hartytp, 2019-06-13:
@gkasprow thanks for doing that! I'm reviewing now.

hartytp, 2019-06-13:
@gkasprow I've completed my review of the main board.

I found more issue than I'd hoped. I was trying not to raise any cosmetic issues -- all designs can be improved further, but at some point one has to be happy with a working design -- however given that we've had some reliability issues in the past I felt it was important to double check the abs max ratings, pin descriptions and typical applications of all ICS in the design. When I did that, I found some issues that look quite important and need to be fixed unfortunately.

hartytp, 2019-06-13:
@gkasprow the schematics are really nice now. Very clean and easy to read! Thank you

hartytp, 2019-06-13:
Finished sup control PCB

hartytp, 2019-06-13:
Completed my review of Booster. Didn't spot any issues with the RF modules.

hartytp, 2019-06-13:
@gkasprow is this change list complete? If so, I think that there are no changes to the supplemental control board or RF modules apart from the ones that Creotech have already fixed on our v1.3 boards. So, I think that if we want to fix all known bugs at some point in the future, we can do that by just swapping out the main PCBs and not touching the RF modules. Is that correct?

gkasprow, 2019-06-13:
that's correct.  We maintain compatibility between v1.3 and v1.4 .

gkasprow, 2019-06-13:
- changed IC9 range from 2.5V to 3.3V (#219), @wizath please change it in the software
- changed current measurement range for 5V and 30V rails, @wizath please change it in the software

hartytp, 2019-06-14:
@gkasprow does the v1.4rc2 release include all these changes? If so, I'll review it now.

hartytp, 2019-06-14:
Broke the software changes out to a separate release to make them easier to keep track of.

closed 2019-06-14 by hartytp

hartytp, 2019-06-14:
@gkasprow okay, I'm done! I've started at the schematics enough that if there are any more mistakes then I can't see them anymore.

In any case, our Booster v1.3 seems to work very well after hand-patching a few things so I have no reason to expect issues.

For the record, the things I've checked in my review are: for every (!) IC in the project I looked at the abs max ratings, pin descriptions and typical applications. One thing I didn't check at all were the passives beyond the information on the schematic. e.g. I didn't look at current ratings for inductors/FBs, voltage ratings for caps etc

hartytp, 2019-06-14:
Thinking out loud, I'm pretty happy with the design now. I think it's generally very good. There are maybe a few things we could simplify now the design has evolved, but nothing that seems worth the effort it would take.

The one thing I would reconsider if I started again from scratch is the pre-amp gain. In hindsight, 40dB is probably a little high. e.g. it means that 0dBm is the max input to saturate the amplifier. However, Urukul can put out more like 10dB. Putting in a lower input signal makes the broad-band noise worse (since we're not far off the thermal floor). Also, scrapping a pre-amp stage would save power and reduce complexity.

That said, I don't care enough about this to want to touch the RF circuit! Just thinking out loud...

Anyway, good job all, thanks again for the hard work. This is a really lovely project now. Time to get back to using it to do physics \o/

gkasprow, 2019-06-14:
OK, So I'm fixing the inamp issue and will release production files.

hartytp, 2019-06-14:
PS other thing to note: my design review was purely on the schematics. I didn't look at the layout at all. I don't have any reason to suspect layout issues, but I didn't check either...

gkasprow, 2019-06-14:
Tom, It was an exciting adventure!
There were so many emotions, right and wrong decisions (mainly related to human interactions at the WUT).
Sometimes I had an impression that your reserve of patience is nearly endless :)
But frankly speaking, it was a challenging project which involves lots of engineering: mechanics, thermal, RF, and analog electronics.
If I started it again, the design wouldn't be that different from what we achieved, which means the design is excellent!

hartytp, 2019-06-14:
Indeed...an adventure with many emotions (including my wife wondering why I’m on git  hub all night) :)

Glad to be freeing up some time for new projects now.