sinara-hw/Booster #213: InAmp supply

hartytp, 2019-06-13:
@gkasprow those InAmps are not capable of operating beyond their supply rails however they are currently powered from 3V3.

Can we supply them both from the 30V rail?

hartytp, 2019-06-13:
not to self: double check the InAmp gain calculations

gkasprow, 2019-06-13:
True, I simply replaced other ones but their architecture is totally different!
Good catch. And the gain setting resistors should be on negative input.

closed 2019-06-13 by gkasprow

hartytp, 2019-06-13:
Aah good catch!

gkasprow, 2019-06-13:
I supplied IC6 from 30V, IC7 from 5V.

hartytp, 2019-06-13:
Okay, that's fine if it makes the routing easier. However, the 5V supply shuts down the InAmp output isn't valid any more, so the diagnostics become unreliable. That can be a little confusing for the users. That's why I'd prefer to power both InAmps from a supplies that aren't shutdown when the modules are off.

But, don't worry about that if it makes the routing much harder.

hartytp, 2019-06-13:
Note to self: checked the InAmp minimum output voltage. Typical (max) is 10mV (65mV). So, with our gain RTI that's about 0.15/1mV. So, basically comparable to the offsets.

tl;dr these diagnostics are only good to 10%, but that's fine for this application.

hartytp, 2019-06-13:
Note to self: checked the InAmp minimum output voltage. Typical (max) is 10mV (65mV). So, with our gain RTI that's about 0.15/1mV. So, basically comparable to the offsets.

hartytp, 2019-06-14:
@gkasprow note that the way the InAmps are connected, they measure the current drawn by the InAmp as well as the current drawn by the amplifier. Normally, I would have connected their supply to the other side of the sense resistor, however in this case I think it's fine:

- the InAmp output current is 1000 times smaller that the current they sense (Rsense/Rset) and the total quiescent current they draw is only 0.15mA. So the error in the current measurement is tiny even for a 50mA load current.

gkasprow, 2019-06-14:
True. Let's leave it and produce as it is.
I will fix it if you find some other serious issue.