sinara-hw/Booster #215: IC8 vset capacitor

hartytp, 2019-06-13:
@gkasprow why have you used an electrolytic capacitor here? It's quite a high-impedance node aren't you worried about leakage currents in the capacitor? The chip is designed to use a ceramic one I believe

gkasprow, 2019-06-13:
Ceramic ones pickup the noise due to piezo effect. We want to keep it low.

hartytp, 2019-06-13:
Aren't you worried about leakage current then? The set-point current is only 100uA.

gkasprow, 2019-06-13:
That's true. The leakage current for TCJA caps is roughly 4uA per each 10uF a 4V.
Popular electrolytic ones have a leakage current of roughly 0.01CV which in our case it would be 0.3uA.
So if you don't mind, let's replace it with electrolytic cap.

hartytp, 2019-06-13:
Okay 0.3% leakage is fine. That way the temp co will easily be dominated by the FETs. That sounds good then

closed 2019-06-13 by gkasprow