sinara-hw/Booster #216: main board R55

hartytp, 2019-06-13:
The 3V3 ref LED current set resistor is 10k. That won't be very bright will it? Shouldn't this be 470R or something?

Also, with that LED do we need R47 at all?

hartytp, 2019-06-13:
So I think we can scrap the annotation about initial load for the LT3042

hartytp, 2019-06-13:
hmm...a lot of the LED resistors are 10k. Are these very low current LEDs?

gkasprow, 2019-06-13:
Yes, these are debug LEDs, not visible foe the user..
First version used 1mA and it was too bright.

hartytp, 2019-06-13:
Okay, up to you. If you're happy with the current resistors then please close. Otherwise change the value to something acceptable.

Either way I'm happy (I was just a bit surprised when I first looked at the schematic).

gkasprow, 2019-06-13:
I changed it to 3k9 to adjust brightness to be similar as other LEDs.

closed 2019-06-13 by gkasprow