sinara-hw/Booster #217: IC8 EN

hartytp, 2019-06-13:
why is this run from 3V3 rather than 5V0? Abs max is 22V so there is no issue driving from 5V0. Reasons I'd prefer to drive from 5V0:
- routing is easier (just tie pins 1-3 together)
- no possibility of EN going high before Vin, causing current to flow through protection diodes
- 3V3 is quite noisy (just SMPS output) so more chance for noise to couple

gkasprow, 2019-06-13:
OK, I wanted to make sure the 3V3_REF won't wake up before P3V3 because I planned to connect it also to VDD of CPU ADC, but it was not easy to route it. Let's connect it to 5V0.

hartytp, 2019-06-13:
> OK, I wanted to make sure the 3V3_REF won't wake up before P3V3 because I planned to connect it also to VDD of CPU ADC, but it was not easy to route it

Aah, I see. 3V3_REF does not go to the CPU ADC does it (in any case I don't think I'd want that, as I don't want anything switching from this ref)? So, I don't think that's an issue.

closed 2019-06-13 by gkasprow