sinara-hw/Booster #219: ADS7924IRTET supply

hartytp, 2019-06-13:
@gkasprow I've just noticed something: the ABS max ratings of the ADS7924IRTET say that AVDD must be >= DVDD (table 6.1). We use AVDD = 2V5 and VDD = 2.5 so we are outside of the max ratings and liable to have component failure. Do we need to use a 3V3 reference for this (not the same one as the bias current since I don't want to put glitches on the FET bias when the ADC samples).

Sorry for not catching this before, however it's a serious issue that must be fixed.

hartytp, 2019-06-13:
Sounds good. Please don't use the same 3V3 ref we use for the bias as I want to keep that very low noise.

hartytp, 2019-06-13:
Can you get away with using the P3V3 SMPS and an RC filter as the reference here?

This is only used for diagnostics, so the accuracy doesn't have to be so good. So long as we can measure the bias current to a few % I'm happy.

gkasprow, 2019-06-13:
OK, this will simplify the thing. I added 10nF, 1uF and 220R FB

hartytp, 2019-06-14:
It might seem

hartytp, 2019-06-14:
> I added 10nF, 1uF and 220R FB

Needs a 1k resistor here. The SMPS noise is too low frequency for a FB to have any affect, I'm afraid.

hartytp, 2019-06-14:
It might seem odd to use a SMPS supply rail as the voltage reference for an ADC. Particularly when that ADC is part of a critical diagnostic (viz the bias current).

Here is my rationalization for why it's okay:
- The level of accuracy we want on the bias current is about 5%. This number is chosen on the basis of:
  a. The InAmp is only 5% accurate because of its offset (typically 150uV on a 5mV sense signal with 50mA bias and 0.1Ohm Rsense)
  b. The FETs are quite temperature sensitive so the required calibration is temperature sensitive anyway. We did originally plan to feed forward on the bias voltage based on the device temperature, but didn't bother because...
  c. The device performance isn't particularly sensitive to bias current anyway and I don't think a 5% change in the bias current will make any noticeable difference to the RF properties
- To get 5% accuracy on the measurement, the reference voltage only needs to be stable to 5% (all noise is scaled by the ADC) 5% of a 3V3 SMPS is 165mV. 
- By the time the converter has gone through an aggressive RC filter, the noise won't be an issue at all.
- The converter accuracy and regulation should be better than 5%.
- NB the InAmp output is 50mA*0.1Ohm*43V/V=215mV. So, 5% is 11mV, which is still quite large

So, I think this is fine, particularly since it makes the rougint so much easier!

gkasprow, 2019-06-14:
IAVDD is max 8uA so let's stick with 1k resistor instead of FB. The FB is essentially 20mili Ohm resistor for such frequencies, but it is still able to eliminate glitches.

closed 2019-06-14 by gkasprow