sinara-hw/Booster #220: ADS7924IRTET input resistance

hartytp, 2019-06-13:
@gkasprow the data sheet for this ADC states that it requires either a low imput impedance or an OpAmp between ADCIN and MUXOUT to function correctly. See figure 58.

Is it okay that we drive the ADC with no buffer and a 10k source impedance? In particular, this means that the 100R filter (R56) is pointless since the pole freq is set by the input resistance.

This explains why @wizath had issues with glitches and had to set a very low ADC sample time.

gkasprow, 2019-06-13:
This mode of operation is visible on [fig 59](http://www.ti.com/lit/ds/symlink/ads7924.pdf)
The ADC needs low impedance, not resistance. That's why 100nF caps are added at its input.

hartytp, 2019-06-13:
Okay, so long as you're happy with the 100nF capacitor output impedance then that's fine.

My concern was that if you drive the ADC with a high DC impedance then you may find that the value you measure depends a lot on the sampling duty cycle (the sampling capacitors move charge between the input capacitors). If the duty cycle is low this won't be an issue.

If you're happy with the current design please close.

gkasprow, 2019-06-13:
we are sampling it very, very slow. So the voltage has enough time to rebuild.

closed 2019-06-13 by gkasprow