sinara-hw/Booster #230: 5V supply limit

hartytp, 2019-06-13:
@gkasprow on the main PCB schematic it gives the max P5V0 current as 250mA. However, the two pre amplifiers have max currents of 180+122=302mA.  The regulator can handle much more power than this, so why is the annotation there?

hartytp, 2019-06-13:
Note that the power budget on the main board is wrong. I believe the total current 6V5 current draw can be quite a bit bigger than 2A. Is there enough margin in all the supplies?

hartytp, 2019-06-13:
Well, I guess the number on the design is about right for the typical power draw. If this is just an estimate of the typical powers then that's fine. If it's actually the max possible rail power then it's not okay. Can you just make it clear which it is, and check that we do have plenty of supply margin.

gkasprow, 2019-06-13:
the IC19 which is the source of the power for 5V LDO has max 3A of current capability. We have also CPU and logics supplied from this rail (2.6W max) + 5V rail (1.2W) which give another 0.6A
We have 8 modules so each can consume 3-0.6 / 8 = 325mA if we assume no margin. 
So we stay within a safe limit. I will update the annotation.

closed 2019-06-13 by gkasprow