sinara-hw/Booster #232: InPwr ADC

hartytp, 2019-06-13:
@gkasprow this is more of a question than an issue...when the input power detector ADC samples, will that put a large glitch on the line, potentially triggering the interlock? The OpAmp output impedance is now 100Ohm and there is quite a bit of trace inductance.

If the signal only changes slowly and we sample the ADC frequently is there a glitch at all? Or does the sampling capacitor retain its charge so that it doesn't produce a spike each time it's sampled?

Anyway, I guess the real question is whether you're happy that the power detector ADCs (input, reverse and output) are all driven by a suitably low impedance?

gkasprow, 2019-06-13:
The inputs of ADC are blocked by 3.3nF caps. The input cap of the S&H in ADC is the order of single pF. So we won't observe it.

hartytp, 2019-06-13:
Not the I2C ADC on the supplemental control pcb. That has no cap

gkasprow, 2019-06-13:
OK, let's add RC then. 1k and 100n should work.

hartytp, 2019-06-13:
1.6kHz pole freq. So that makes the detector pretty slow, but I guess the I2C ADC polling is very slow already so that's not a problem!

gkasprow, 2019-06-13:
Exactly.

closed 2019-06-13 by gkasprow

hartytp, 2019-06-14:
@gkasprow The only other place this sampling cap issue arrises is the microprocessor. Do you think it's an issue that there are no sampling capacitors there? There is quite a bit of trace inductance on top of the FBs etc, so the source impedance will be high. 

On top of that, is there a risk that the ADC sampling will put glitches on the power detector measurements that could cause the interlocks to trip accidentally?

I don't think this is a problem in practice (and, in any case, the AC hysteresis prevents the interlock from tripping on a short glitch) but I was interested to hear your view...

gkasprow, 2019-06-14:
There are 3.3nF caps at the CPU ADC inputs.  They are placed close to the DSUB connector.

hartytp, 2019-06-14:
Aah, yes, I'd forgotten those! Thanks for the reminder. In that case, that's fine. Good job we added that OpAmp output isolation resistor, as it wouldn't have liked driving that capacitance directly.

hartytp, 2019-06-14:
Note to self: those 3.3nF capacitors in combination with the OpAmp output resistance give a few hundred ns 1/e rise time. That's fine since we only want the interlock to trip on a timescale of a few ns (we don't want it to trigger from very short spikes due to RF switches etc)