sinara-hw/Booster #236: InAmp connections

hartytp, 2019-06-14:
@gkasprow unless I'm going crazy (quite possible) I think the InAmps are still connected incorrectly. I think you noticed this at some point during the previous design review, but it isn't fixed yet.

See the schematic on the first page of the InAmp data sheet: the InAmps attempt to make their two inputs equal by sucking current away from the negative input. For this to work, the negative input must be connected to the high side of the sense resistor.

gkasprow, 2019-06-14:
True. For some reason, I assumed that current flows from left to right side. I did these changes late in the night. THat's why I connected amps power supply in such way

hartytp, 2019-06-14:
No problem. I probably wouldn't have spotted it now if you hadn't pointed it out to me yesterday!

gkasprow, 2019-06-14:
Please check with second pair of eyes:
[image]

closed 2019-06-14 by hartytp