sinara-hw/Booster #237: scrap input power intelock

hartytp, 2019-06-25:
@gkasprow What do you think about scrapping the input power interlock completely (both HW and SW)?

- [ ] remove from firmware
- [ ] for now DNP all components associated with it and make sure that couplers are terminated.
- [ ] in some future design revision we can remove it completely if we want to to simplify the design (remove a directional coupler etc)

hartytp, 2019-06-25:
The problem with the input power interlock is this:
- the very fast interlocks (ns response) didn't work since they always tripped from transients. Anyway, we never made the internal comparators work correctly and the RMS output doesn't respond that fast
- so, we now have interlocks that trip in a few us. However, if the interlock only trips in a us or so then there isn't much point interlocking the input as opposed to the output
- currently the input interlock doesn't work anyway since there is a race condition when the DFF is reset (this is because the input interlock is before the RF switch, so the power doesn't go low when it trips)

In other words: with the current design the input interlock is pointless complexity.

So, we have two options:
  a) Fix the input interlock by using the power detector's internal comparator etc. This could be quite some work
 b) Scrap the input interlock and rely on the output interlock to protect the amplifier.

I think we should go for (b) -- scrap the input interlock -- unless we think it's really necessary. Will 0dBm applied for a few us damage the amplifier? If not then why do we need this interlock?

hartytp, 2019-06-25:
@gkasprow what do you think? Can we scrap the input interlock?

gkasprow, 2019-06-25:
Yes, keep the things as simple as possible. a few us won't kill the input.

hartytp, 2019-06-25:
Okay good. I’ll break out into separate issues

closed 2019-06-25 by hartytp