sinara-hw/Booster #242: Bug: No Set-Command Response

pathfinder49, 2019-07-18:
The wiki states the following about Booster coomand responses:
> Booster responds to all commands. Commands containing a `?` query Booster for information, an error string is returned in case of a programming error, such as a malformed command. Successful command without any return value should return just a new line character (\n).

However, Booster does not respond to commands not ending in `?`.

The following **python 3** code demonstrates the issue:

```Python
import serial

s = serial.serial_for_url("socket://10.255.6.249:5000", timeout=0.1)

cmd_list = ["*IDN?\n",  # this command is not documented

			"chan:enab 0\n",  # should return b"\n"
			"chan:enab? 0\n",  # should return b"1\n"
			"chan:disab 0\n",  # should return b"\n"
			"chan:enab? 0\n",  # should return b"0\n"
			]

for i, cmd in enumerate(cmd_list):
	s.write(cmd.encode())
	dump = s.read(1024)
	print("dump {}:".format(i), dump)
```
Console:
```
dump 0: b'RFPA v1.4.19165 3df8da0, built Jun 14 2019 12:45:01, hv rev 3\n\r\n'
dump 1: b''
dump 2: b'1\r\n'
dump 3: b''
dump 4: b'0\r\n'
```

hartytp, 2019-07-30:
@wizath when are you going to fix this?

We agreed on a clear specification for how the remote interface should behave. This included Booster sending a response to acknowledge all commands. I have asked you multiple times to check that the firmware *actually does what it says on the wiki*, so it's unacceptable that it still does not.

Fix this and the other firmware issues asap, I don't have any more time to waste on silly firmware bugs.

wizath, 2019-09-03:
Same as https://github.com/sinara-hw/Booster/issues/244. Fixed.

closed 2019-09-03 by wizath