sinara-hw/Booster #243: Bug: Booster Identification Response

pathfinder49, 2019-07-18:
The Booster command (`b'*IDN?\n'`) returns an additional new line character.
I receive the following response to the command: `b'RFPA v1.4.19165 3df8da0, built Jun 14 2019 12:45:01, hv rev 3\n\r\n'`
Presumably the first `\n` character shouldn't be present.

Further, this command is not documented in the wiki.

hartytp, 2019-07-30:
@wizath This is really obnoxious because it means that my python code has to do this
```python
    def identify(self):
        """ Returns a device identification string """
        self.dev.write(b"*IDN?\n")
        idn = self.dev.readline().decode().strip()
        self.dev.readline()  # flush spurious new line character
        return idn
```

if I don't have the second readline then the next command I send reads a blank response.

wizath, 2019-07-30:
fixed

pathfinder49, 2019-07-30:
Thanks! However, the identification command is still bot documented in the wiki. It definitely should be documented.

wizath, 2019-08-14:
Command definition is now listed in SCPI command set. [Link](https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface#idn)

closed 2019-08-14 by wizath