sinara-hw/Booster #244: Documentation: Booster Querie Responses

pathfinder49, 2019-07-18:
Booster responses seem to be terminated with b'\r\n'. This behaviour should be documented in the wiki.

```Python
import serial

s = serial.serial_for_url("socket://10.255.6.249:5000", timeout=0.1)

cmd_list = ["*IDN?\n",  # this command is not documented

			"meas:temp? 0\n",
			"chan:enab? 0\n",
			"not_a:command 0\n",
			]

for i, cmd in enumerate(cmd_list):
	s.write(cmd.encode())
	dump = s.read(1024)
	print("dump {}:".format(i), dump)
```
Console:
```
dump 0: b'RFPA v1.4.19165 3df8da0, built Jun 14 2019 12:45:01, hv rev 3\n\r\n'
dump 1: b'23\r\n'
dump 2: b'0\r\n'
dump 3: b'[scpi] **ERROR: -113, "Undefined header"\r\n'
```

pathfinder49, 2019-07-18:
I'm happy to update the wiki if I can get edit permissions.

wizath, 2019-09-03:
That `\n` fooled me while I was using ncat. With python commands that don't produce output does not return anything. I've changed response behavior to return:
* value or list of values if command is supposed to return anything
* error string if something goes wrong
* 'OK' string if command is executed successfully and does not return anything (e.g `chan:enab 1`)

closed 2019-09-03 by wizath

hartytp, 2019-09-03:
Perfect, thanks!