sinara-hw/Booster #245: Hardware interlock hysteresis

RHanley1, 2019-07-25:
I am currently testing a Booster v1.3. There appears to be hysteresis with regards the hardware interlock tripping and resetting. 

For example, the output interlock is set to trip at 30dBm. The interlock correctly trips at 30.14 dBm. However, to reset the interlock, I have to reduce the RF input power such that the output RF power is ~28dBm before the interlock can reset.

This makes the interlock feature rather difficult to use if one is operating near to the interlock threshold.

hartytp, 2019-08-05:
@wizath did you see this? I don't understand where this effect comes from. Unless you can see an obvious cause of this, when we received another v1.3 booster I will set up some probes to look for glitches etc on interlock resets

hartytp, 2019-08-06:
I strongly suspect that this is a hw issue with the same cause as this https://github.com/sinara-hw/Booster/issues/258#issuecomment-518668185

@wizath when the new boosters arrive, please can you check that this no longer happens? Thanks!

hartytp, 2019-09-04:
Hysteresis is now about 0.3dBm-0.4dBm. Not sure exactly where the residual hysteresis comes from, but this doesn't seem to be an issue.

hartytp, 2019-09-04:
:fireworks: 

so...for the first time...there are no **known** hardware bugs. Just need some final fw fixes and we're good to go! Whoo!

closed 2019-09-04 by hartytp