sinara-hw/Booster #246: Hardware interlock tripped by power measurement

RHanley1, 2019-07-25:
I am currently testing a Booster v1.3. There appears to be a problem whereby the use of the MEASure:OUTput? command causes the hardware interlock to trip. 

I can approach the interlock set-point by varying the RF input amplitude, and the interlock trips in roughly the correct place. However, when the interlock is close to tripping, but hasn't actually tripped yet, and one uses the internal power meter MEASure:OUTput? command, it causes the hardware interlock to trip.

hartytp, 2019-07-30:
@wizath what is going on here? Why does querying the output power over the remote interface trip the interlock?

wizath, 2019-07-30:
Can't really reproduce this issue on my device. Could you post detailed steps?

hartytp, 2019-07-30:
Okay, I'll have another look at this.

RHanley1, 2019-08-07:
Can't seem to reproduce this myself anymore either. Best to close this issue for now.

closed 2019-08-07 by RHanley1