sinara-hw/Booster #248: input validation

hartytp, 2019-07-30:
basically works, but doesn't catch things like setting `channel = 7.1`

hartytp, 2019-07-30:
Note to self, things I've checked on a few random commands:
- channels are validated (modulo above point) correctly
- unexpected arguments raise errors

hartytp, 2019-07-31:
seems to work well.

hartytp, 2019-07-31:
oops, forgot the fractional channel issue, which does need fixing

wizath, 2019-09-03:
added errors for
* channel out of bounds (0-7)
* empty channels (not detected)
* wrong interlock values (0 - 38 dBm)
* too many parameters
* confirmation "OK" for commands that don't return anything

Checked most of commands for these errors, seems to work fine

hartytp, 2019-09-10:
Fractional values still uncaught!

wizath, 2019-09-10:
Where?

hartytp, 2019-09-10:
> *IDN?
>> rfpa v1.4.19248 2d079d8', ' built sep  5 2019 14:57:58', ' id 30374705', ' hv rev 4
> CHAN:ENAB? 1.1
>> 1

wizath, 2019-09-10:
Ok, I think I've got line causing issue

wizath, 2019-09-10:
That's related to commands that can take ALL parameter, will fix when I'll get to computer

wizath, 2019-09-10:
Interesting, I've tested other commands for double/int pareter parsing but not these commands

wizath, 2019-09-11:
It turns out that SCPI library just takes string with parameter and then passes it into clib `strtol` or `strtod` functions. That function does not care if input is `20.0` or `20` it will make it proper format. 

I've added checking if string contains `.` to detect if it's double and then reject the value.

hartytp, 2019-09-11:
> I've added checking if string contains . to detect if it's double and then reject the value.

Might be better to add a strict test for only characters 0-9, +- so that we also take out silly cases like `1e1`

wizath, 2019-09-11:
There is validation for correct channel later - only accepts 0-7

wizath, 2019-09-11:
Added checking for both floats and `1e1` numbers

hartytp, 2019-09-11:
can you push the latest code?

closed 2019-09-11 by hartytp