sinara-hw/Booster #249: Inconsistency between on-board detector and interlock

AUTProgram, 2019-07-30:
The readings returned from the on-board power detector via get_output_power() are not consistent with the interlock threshold. For a given interlock setting, the power detector measures values  that can be +-0.7 dB higher or lower for some channels at the point when the interlock trips.
So when I set the interlock to 30.0 dBm for example and then increase the RF power applied to the input in 0.1 dBm steps while repeatedly reading the output via get_output_power(), for some channels the interlock trips at a measured value of 29.3 dBm. For others the detector measures up to 30.7 dBm when the interlock trips.

hartytp, 2019-08-05:
I suspect that the cause of this is the fact that Creotech used DC hysteresis for the interlock instead of the AC hysteresis we requested. This moves the interlock voltage relative to the voltage measured by the CPU ADC, which is unfortunate.

To check we understand this:
- [ ] @wizath please can you confirm how the power detector and interlock are calibrated? e.g. is there a single calibration for both the on-board detector and the interlock? Or, do they have separate calibrations
- [ ] @hartytp to check over the schematic and see how accurately we expect the interlock setting to matcht he measured power if a single calibration is used (DAC offsets etc)
- [ ] @hartytp check that v1.4 performs as expected

hartytp, 2019-08-06:
I suspect that this has the same cause as this https://github.com/sinara-hw/Booster/issues/258#issuecomment-518668185 @wizath please verify that this isn't present on the v1.4 hw and we can close

wizath, 2019-08-14:
v1.3 have separate calibrations for power-monitoring and interlock settings. v1.4 is delayed a little because of hardware mistake. I will test it with single calibration to see if it behaves better.

hartytp, 2019-08-14:
Thanks for clarifying.

In the v1.3 hardware we have the interlock and power detectors show significant discrepancies, which is odd if they were both calibrated. Eg if we set the interlock to 30dB then it does not trip when the detector reads 30dB. Maybe this is due to the hysteresis?

Anyway, yes let’s check this again on the v1.4 hardware and hopefully the effect will be gone :)

hartytp, 2019-08-14:
Thanks for all the work you’ve done on Booster recently!!

hartytp, 2019-09-04:
Quick test on a single channel of one of our Booster v1.4s. 200MHz RF, interlock set at 30dBm. Interlock trips when input power detector is at 30.0(1)dBm. Nice!

closed 2019-09-04 by hartytp