sinara-hw/Booster #251: more helpful errors

hartytp, 2019-07-30:
setting an invalid interlock threshold gives the error "b'[scpi] **ERROR: -200, "Execution error"\r\n'" this is not very helpful, and will make debugging hard. Can we have slightly more descriptive errors (we don't have to stick rigidly to the SCPI standard for this)?

wizath, 2019-09-03:
Added some more meaningful errors
`[scpi] **ERROR: -99, "Channel not detected"`
`[scpi] **ERROR: -98, "Interlock value invalid (0-38 dBm)"`
`[scpi] **ERROR: -97, "Wrong channel selected (0-7)"`

hartytp, 2019-09-03:
Reopening in case you weren't finished. If you've implemented all the errors you think are important, please close.

wizath, 2019-09-05:
There is also an old one in case of excessive amount of parameters 
`[scpi] **ERROR: -108, "Parameter not allowed"`.

 I think all error cases are now properly labeled.

closed 2019-09-05 by wizath

hartytp, 2019-09-05:
Cool, thanks! Good errors make the user experience so much better.