sinara-hw/Booster #252: booster sometimes stops responding to commands

hartytp, 2019-07-30:
During testing, I've noticed that sometimes sending series of commands like:
```
CHAN:DISAB all
MEAS:CURR? all
CHAN:ENAB all
MEAS:CURR? all
```
results in booster not responding to one of the current measurement requests. If I leave a long delay between commands then there doesn't seem to be an issue, but sending the commands close together seems to cause issues.

NB since booster currently does not respond to the `CHAN:DISAB all` commands, my driver does not know how long to wait between requests...

wizath, 2019-07-30:
By default, you should wait for around 5ms between each request

hartytp, 2019-07-30:
What do you expect to happen if multiple requests are received simultaneously? Silently dropped commands seems like a very bad thing...

I was waiting ~100ms between requests when I saw  this behaviour

wizath, 2019-09-03:
I think that is related to https://github.com/sinara-hw/Booster/issues/265. I did not observe such behavior after fixing that issue.

hartytp, 2019-09-03:
Thanks! I'll re-test and open a new issues if I see further issues

closed 2019-09-03 by hartytp