sinara-hw/Booster #255: INT:STAT v INT:OVERLOAD

hartytp, 2019-07-30:
what's the difference? Not clear from the descriptions on the Wiki.

hartytp, 2019-07-30:
Similarly, for `INT:ERROR?`

hartytp, 2019-07-30:
Side note: I've not done as much testing on Booster as I can with the current state of things. I've reported all bugs I've found.

I need all the open fw issues fixed asap so I can finish testing. @wizath when will you do this?

wizath, 2019-08-14:
Ok, fixed the wiki descriptions.

`INT:ERROR?` - return the error status (red light on FP, like bad temperature readings, communication error)
`INT:STAT?` - return 1 if either input our output interlock will trip
`INT:OVERLOAD?` - return only output interlock status

Close if solved

hartytp, 2019-08-14:
Thanks. What about the reverse power interlock? Also we will remove the input interlock. Maybe 4 commands: error, forward reverse and status (status is the combined interlock and error status so you can quickly check that everything is okay). What do you think?

wizath, 2019-09-03:
* STATus - output interlock, reverse interlock and error status
* FORward - output interlock only
* REVerse - reverse interlock only
* ERRor - error condition

- [ ] Add missing commands to wiki

closed 2019-09-03 by wizath