sinara-hw/Booster #256: report hardware unique ID as part of the version string

hartytp, 2019-07-30:
The string returned by the version command must contain some data that can be used to uniquely identify the hardware to the manufacturer. I assume this will be done by MAC address (and Creotech/Technosystem will take note of the MAC at production time).

wizath, 2019-08-12:
ATM I'm using CPU UUID for identification. From VCP you can use `devid` to get device unique ID and `chanid` for specified channel. Do you want it added to `*IDN?` command or get another one that returns only the ID?

hartytp, 2019-08-12:
What I want is an easy way of getting a unique device ID via SCPI. This is useful e.g. if there is an issue with the hardware that I want to discuss with the manufacturer. I don't mind whether that's the CPU UUID or MAC, so long is it's something the manufacturers can use to Identify the device.

> Do you want it added to *IDN? command or get another one that returns only the ID?

Ideally, I'd like the `*IDN?` command to return a string with: HW ID, HW design rev, firmware ID (e.g. commit), firmware revision/build date. So something along the lines of (feel free to use whatever you want so long as it has this info included) "Booster RF PA hw rev v1.4; device ID 0x000; firmware v0.0 (git hash) build date ".

wizath, 2019-08-19:
`*idn?` now returns
`RFPA v1.4.19231 7e96754, built Aug 19 2019 17:33:16, id 34364708, hv rev 3`

closed 2019-08-19 by wizath