sinara-hw/Booster #258: booster bistability

hartytp, 2019-07-31:
@gkasprow I've observed a small bistability on Booster's power detectors a few times. 

I'm running Booster from Urukul at a constant power, frequency and phase. I'm logging all the Booster diagnostics using the VCP interface. The observation is that sometimes the power changes by ~0.3dB and the noise changes at the same time. The only correlation I can spot in the diagnostics is that the 5V MP voltage changes a bit at the same time.

See this screenshot:

[image]

@gkasprow can you think of a cause of this??

Next I'll add an external power meter to log the output power and see if this is an issue with the Booster diagnostics or if it's actually on the output...

hartytp, 2019-07-31:
cc @cjbe

hartytp, 2019-07-31:
Another screenshot showing a wider time window (including the period before I turned Booster on). The variations in the 5VMP rail are interesting...

[image]

hartytp, 2019-07-31:
Side note, the stability of the Urukul/Booster system is really quite good. Here is a log that starts just after I switch Urukul on. You can see the drift/thermal transients are barely measurable when using an external detector (V3500A). Pretty good!

[image]

hartytp, 2019-08-02:
The 5V0 MP powers the interlocks and diagnostics (thermometer and power detectors). So the correlation between this noise/step change in power and the 5V rail suggests that this is an issue with the diagnostics rather than the amplifier itself (although I'll know for sure once I reproduce this while monitoring Booster's output with an external detector).

25mV is a large change in the regulator's output, so suggests that something is really quite wrong. Maybe an oscillating IC or something???

hartytp, 2019-08-02:
NB the measurement of the 5V rail is after the fuse, which has 100mOhm of resistance or so IIRC (to do: check). So, this suggests a large current draw on that rail.

Could also be an issue with the ADC since we're operating outside its abs max ratings. To do: look at the v1.3 - v1.4 diff and see what could cause this.

gkasprow, 2019-08-05:
@hartytp the 5VMP rail was originally meant to supply only EEPROMs, thermometers and diagnostics. It is regulated down to 3.3V so small variations shouldn't make any issues. It supplies also power detectors which are a form of diagnostics as well. The idea was to keep all these circuits operational even when preamp 5V is off.

gkasprow, 2019-08-05:
Why do you mean by "Could also be an issue with the ADC since we're operating outside its abs max ratings" ?

hartytp, 2019-08-05:
The 5V measurement could be suspect since the ADC we use to measure it is operating outside its absolute maximum ratings IIRC (this was one of the issues we fixed in v1.4)

hartytp, 2019-08-05:
@wizath when you're back in, please can you look at #262 first. What's not clear yet is whether this is an issue with (a) Urukul (our source) (b) Booster as an RF amp (c) Booster's diagnostics. I have set up logging with an external detector to distinguish between (b) and (c), but the internal input detector would allow me to separate (a) from (b) easily

hartytp, 2019-08-05:
> @hartytp the 5VMP rail was originally meant to supply only EEPROMs, thermometers and diagnostics. It is regulated down to 3.3V so small variations shouldn't make any issues. It supplies also power [...]

@gkasprow I know. But don't you find it odd that the voltage on this rail suddenly changes by so much? Maybe a coincidence but it looks like there may be a correlation between the shift in level on the 5VMP rail and the output power measurement.

If the effect is real I would guess the causation is the other way around: whatever is causing the measurement change is also altering the current draw and hence changing the rail voltage (i.e. it's not the rail voltage changing the reading I'd guess).

hartytp, 2019-08-06:
@gkasprow I can definitely use the 5VMP rail as a crude power detector. When I turn the RF on to around 1W the 5VMP rail changes by approx 5mV. Not a lot, but definitely noticeable.

This is a pretty small effect and could just be ground potential changes due to the 30A current or something like that. If it were real and we assume that the impedance of this rail is dominated by the fuse, then the change in rail current would be dI = 5e-3V/0.3Ohm = 15mA. That feels like quite a large change! 

Fuse data sheet [here](http://www.farnell.com/datasheets/2282569.pdf?_ga=2.189691051.103083363.1565082009-246918782.1551950583&_gac=1.204262948.1565082009.EAIaIQobChMIgfKgz_Dt4wIVRrDtCh03fgLhEAAYAyAAEgI8lvD_BwE) @gkasprow what is `R1max`?

hartytp, 2019-08-06:
Based on https://github.com/sinara-hw/Booster/issues/192#issuecomment-496306659 and https://github.com/sinara-hw/Booster/issues/158#issuecomment-496033224 I believe that Creoteh used a 1k resistor for the DC comparator feedback. The pick-off (including attenuator) is -20dB. So, at 1W output, the power detector output is something like 1.5V. Offset is small so we can ignore. So, the OpAmp output voltage is approx 3V. OpAmp output voltage is then approx 2.25V. So, the we expect an increase on the 5VMP rail of approx 2.25mA just from the comparator load. 

However, a much bigger effect is the power detectors themselves. The increase in current for 10dBm input at the detector is around 10mA. So, taking these effects together, this change in 5VMP supply voltage with output RF output power is not surprising or worrying...

[image]

hartytp, 2019-08-06:
aha!

@wizath @gkasprow look at this:

[image]

Here, I'm changing the interlock threshold from far from tripping (37dBm) to close to tripping (29.5dBm) and back to far from tripping (back to 37dBm).

Observations are:
- the output power measured by an external power detector does not change (good!)
- the internal output power measurement changes 
- the internal output reflected power does not change
- the 5VMP current draw changes

I believe that what is happening here is that Creotech installed DC feedback. My guess is that when the comparator gets close to switching its DC output starts to rise, which is able to change the voltage on the output of the OpAmp. Or something like that.

@wizath once the v1.4 Boosters come in, please can you verify that this effect is no longer present

hartytp, 2019-08-06:
The nearly15mV drop in the 5V rail current is a little scary! Not totally sure what causes that, but if I had to guess, I'd suggest it's the comparator starting to oscillate, with the load provided at AC by the feedback capacitor. I guess that could sink quite a bit of current into the filter capacitors. Again, this should be fixed with the proper AC feedback, but will check once the new HW arrives.

gkasprow, 2019-08-07:
It's quite possible that your version has not proper low impedance feedback which causes high current flow.

hartytp, 2019-09-04:
Can't reproduce the changes in measured output power as a function of the interlock set point. Noise on power detector is basically ADC quantization-limited even 0.1dB away from the interlock threshold.

So this issue seems fixed

closed 2019-09-04 by hartytp