sinara-hw/Booster #262: reinstate input power monitoring

hartytp, 2019-08-05:
- [ ] add to VCP diagnostics (including data dumps)
- [ ] add to SCPI
- [ ] add to docs

wizath, 2019-09-04:
Checked input power detector variations between channels. This feature would need (quick) calibration in order to provide meaningful (dBm) readings. The characteristic is linear so it's quite easy to do.

[image]

hartytp, 2019-09-04:
@wizath thanks. Yes, I guess this can be calibrated at the same time as the output power detector

hartytp, 2019-09-04:
(I'm happy recalibrating our existing devices once this feature is added)

hartytp, 2019-09-04:
BTW, can I have a binary with the most recent firmware to flash?

wizath, 2019-09-04:
I'll upload it tomorrow when I'll finish testing recent changes.

hartytp, 2019-09-04:
perfect, thanks! I'll aim to test the rest of the fw tomorrow and sign off on it if possible

wizath, 2019-09-05:
[image]

Results after simple input calibration. Not bad I guess

hartytp, 2019-09-05:
Nice!

wizath, 2019-09-05:
Looks like valid measurement range is like > -20 dBm

hartytp, 2019-09-05:
That sounds about right. -20dBm is about +20dBm output. Realistically, if people want to use less than that then Booster is probably not the right amplifier choice anyway :)

wizath, 2019-09-05:
- [x] Added to `status`, and diagnostic table (raw voltage and calibrated power)
- [x] Added to SCPI `MEASure:INput?` (wiki pending)
- [x] Added calibration commands
- [x] Checked accuracy of calibrated measurements

closed 2019-09-05 by wizath

hartytp, 2019-09-05:
great, thanks!