sinara-hw/Booster #263: finish proof-reading the wiki

hartytp, 2019-08-05:
Doesn't need to be a work of literature, but should be understandable and contain all the relevant info.

@dnadlinger did you take a nice photo of Booster we can use?

hartytp, 2019-08-06:
@wizath Calibration says `TODO ADD CLI OUTPUT`

hartytp, 2019-08-06:
``` It'll be fixed soon, but for now, you can just redo your calibration with slightly higher current eg. 52 mA and it should be fine``` is this fixed yet?

hartytp, 2019-08-06:
Is there a calibration for the input power detector? It's not on the wiki...
- [ ] to do: delete input power detector calibration from wiki

hartytp, 2019-08-07:
@gkasprow Nice paper from the Maryland people https://arxiv.org/pdf/1908.02156.pdf (designs at 
https://github.com/JQIamo )

Similar amplifier to Booster. From a quick skim (may contain mistakes!), the comparison is:

- they don't have any of the interlocks/diagnostics that Booster has (which are, IMHO, one of the best things about Booster). e.g. this is why the docs say things like `Do not turn on the Amplifier Power switch without a 50 Ohm load connected to RF Out` (hopefully Booster's interlocks make it extremely robust to this kind of issue). Similarly, bias cal must be done with a trim pot and multimeter rather than a single command and load protection is achieved by putting a fixed attenuator in the box. Power sequencing is an analog circuit controlled by FP switches, rather than DACs + micro-controller with an Ethernet interface.
- they integrate some extra RF bits like VVA, VCO and TTL-controllable RF switch into the amplifier. I still prefer to keep things modular and leave that to something external like Urukul/Mirny, but interesting to see other approaches. I did at one point consider adding an external TTL (e.g. FP MMCX) that can be used to open the RF switch on Booster quickly. With ARTIQ though, it seems nicer to have the switch on an EEM where it can directly connect to the FPGA without a cable
- their design is slightly more expensive (much more expensive PA chip!). They quote $1085 per channel including chassis in quantity whereas our last run of Booster was $900 per channel. Our cost included assembly, calibration, data sheet production etc, not sure if their cost does
- they don't have fans and it seems like the amplifier temperature is marginal (they also don't have thermal interlocks, so this is a little scary)
- their unit has a wider BW up to 1GHz, with lower ouput power (although, I'm not sure if they actually use the top end of the bw for anything in their lab). I actually can't remember what limits the BW of our amp, and whether it's just the matching network design or one of the gain stages.
- the power efficiency of their design is a bit lower than Booster. They need 1A at 24V, IIRC we need something like 0.24mA at 30V running at saturation.
- their design (at least the mechanics) seems much bulkier than Booster (see image at https://github.com/JQIamo/aom-driver/wiki/Front-Panel )

gkasprow, 2019-08-08:
That's a really interesting project! I'm curious about what the current revision it is. It took a few of them to get satisfactory results in most of our projects. If needed, we can always mount an MCX connector with TTL level to every aux board that would protrude through the panel.

hartytp, 2019-09-06:
@wizath I finished reading/tidying up the home page. Can you have a read over it and sign-off on all details being correct to the best of your knowledge. In particular, is this accurate:

> When any interlock trips, the input RF is disconnected from the amplifier using an internal RF switch. In case of an error condition, such as excessive current consumption or temperature in the amplifier, the amplifier power stage is also shut down. Once an interlock has tripped, it must be manually cleared to resume normal operation, either by pressing the "interlock reset" button or via the remote programming interface. Note that disabling a channel (via the remote interface or by pressing the standby button) does not clear the interlock status, despite the fact that the OVERLOAD and ERROR LEDs turn off as long as the channel remains in standby. Once the channel is re-enabled, the interlock LEDs switch on again.

hartytp, 2019-09-06:
firmware update looks good

hartytp, 2019-09-06:
@wizath I read the calibration page and made a few changes to tidy things up. Please can you proof-read and check you agree with it all? In particular, the input power detector calibration process is a little different to the others, since the user does not input the power they calibrate at according to the wiki. Is this correct?

hartytp, 2019-09-06:
Other than that, all that remains are the SCPI and VCP pages, which I'll proof-read while I test once you've finished testing your code.

So...nearly...there...!!

hartytp, 2019-09-06:
hmm...do I really need to drive the input at +3dBm for calibration??

wizath, 2019-09-07:
This was discussed so long ago. You will drive input at 3dBm and output interlock will immediately disconnect the RF switch, but again the input power comparator will still be able to measure the input power (because its placed before RF switch), so calibration was safe to execute in proper order.

Anyway it's now removed

hartytp, 2019-09-07:
Makes sense. Thanks for clarifying. Out of curiosity, how come it’s not a 2 point cal?

What about the couple of other points I raised a few posts up?

wizath, 2019-09-07:
> Out of curiosity, how come it’s not a 2 point cal?  
Because that was fixed value

hartytp, 2019-09-07:
? Not sure I follow. Anyway I don’t mind, just wonder why it’s not the same as the others...

wizath, 2019-09-07:
We removed overcurrent errors from software a while ago since now there is this hardware foldback circuit, so it can be removed from wiki

hartytp, 2019-09-07:
Okay good

Can you read over the wiki and make sure it’s up to date. Then close this issue

hartytp, 2019-09-07:
@wizath really good work on the wiki. Thanks so much!!!

hartytp, 2019-09-10:
@wizath

> We removed overcurrent errors from software a while ago since now there is this hardware foldback circuit, so it can be removed from wiki

This still needs doing AFAICT

Can you read over the wiki and confirm that all details are accurate please? Thanks!

wizath, 2019-09-10:
> This still needs doing AFAICT

Do you mean that we still need overcurrent errors?

hartytp, 2019-09-10:
No, I mean that it needs to be removed from the wiki, since the current Wiki description is wrong.

wizath, 2019-09-10:
Ok, I'm starting reading the whole wiki after my coffe is complete

hartytp, 2019-09-10:
Great!

hartytp, 2019-09-10:
@wizath PS please close this issue when you're finished proof reading and updating the wiki

hartytp, 2019-09-10:
ping...

wizath, 2019-09-10:
ok, for the second time everything seems ok, troubleshooting section looks fine too

hartytp, 2019-09-10:
ace, thanks!

closed 2019-09-10 by hartytp