sinara-hw/Booster #268: Power detector offsets [v1.3 - v1.4]

wizath, 2019-09-03:
[DRAFT]

While testing the v1.4 hardware revision I've noticed higher (than v1.3 averaged) readouts from power detectors. At first glance, there is only a slight difference (averaged from 16 channels each)

[image]

But, when we look on the worst case module from v1.4
[image]

This difference causes interlock calibration to jump out of the 12bit DAC range. In this scenario there is no way to set interlock level higher than 36 dBm (or even to 36 dBm).

[image]

Todo:
- [ ] Don't allow calculated value to overflow DAC
- [ ] Check interlock calibration
- [ ] Redo v1.4 raw detector output characteristics
- [ ] Check schematics for changes

hartytp, 2019-09-03:
@wizath my money here is that the V1.3 power detector voltage was a bit lower than expected because of the DC comparator hysteresis combined with the OpAmp output resistor.

Let's see what we expect:

36dBm output power should correspond to approx 6dBm at the detector

[image]

Power detector output should be a bit below 1.6V
[image]

- OpAmp has gain of 1.5V so 2.4V
- assuming the DC load on the OpAmp is small (it should be!), that's the voltage at the comparator input

The comparator DAC is an [AD5627RBCPZ](https://www.analog.com/media/en/technical-documentation/data-sheets/AD5627R_5647R_5667R_5627_5667.pdf) so full-scale output of 2.5V

[image]

In other words, from the design I'd expect interlocks above around 36dBm to be very marginal. e.g. we start to hit tolerances in RF attenuators, couplers etc. So, I'm not surprised this happens (maybe a better question is why we didn't see this before!).

Do you agree with that diagnosis @gkasprow? If so, it's not hard to fix, but there are a few options. Which is your preferred?

For now, we can live with a max interlock thershold of 36dBm since we never use the full 5W and we can hack some resistor values if we really need to.

Anyway, good catch @wizath!

wizath, 2019-09-03:
I think I'll add method like INT:POW MAX to set interlock level to full 12 bits then

hartytp, 2019-09-03:
> I think I'll add method like INT:POW MAX to set interlock level to full 12 bits then

Very good idea. Implement that and the clipping and I think it's problem solved from a FW perspective.

hartytp, 2019-09-03:
@gkasprow if you agree with my diagnosis then please close this issue and open a new one with your proposed remedy to the issue. Thanks!

wizath, 2019-09-30:
TL;DR

Since after using new comparator hysteresis circuit we are often working at the end of DAC 12-bit range.

Booster_PA_PREAMP -> U7 MCP6291T

gkasprow, 2019-11-05:
OK, so let's make the R31 DNP

closed 2019-11-05 by gkasprow

wizath, 2019-11-05:
R31 is in RET detector, that was a mistake, we need to change FWD detector (R32, R23)

hartytp, 2019-11-06:
> R31 is in RET detector, that was a mistake, we need to change FWD detector (R32, R23) to 1:1 ratio

Sounds good. So we're changing the FWD and RET detector OpAmps to unity gain. SOunds good.

wizath, 2019-11-06:
> Sounds good. So we're changing the FWD and RET detector OpAmps to unity gain. SOunds good.

RET power detector is fine, we need to change only FWD one @gkasprow

gkasprow, 2019-11-06:
ACK