sinara-hw/Booster #270: VCP power measurements not updating correctly

hartytp, 2019-09-04:
On the v1.4 hw we have I just noticed something interesting.
- monitoring the power via the VCP continuous parameter dump (`start`)
- when the interlock tripped for one channel, the reported TX power didn't update
- I used the SCPI interface to query the power. The first time, it still reported a high power
- after running it again, the reported power dropped on both the SCPI and VCP interfaces

After clearing the interlock, I saw a similar thing:
- the VCP still reported a low power
- the SCPI interface also reported a low power the first time I queried it
- after querying the SCPI interface a second time, it updated correctly and the VCP interface started showing a high power again

hartytp, 2019-09-04:
hmm...no this isn't just the interlock. It seems that sometimes the power detector just stops updating in SW. e.g. if I change the RF power the VCP still reports the same power even after 10s or more. The SCPI also shows the previous power the first time I call it, and only updates when I call it a second time. Not sure if a power cycle clears this state.

hartytp, 2019-09-04:
hmm...even updating the interlock threshold seems to cause Booster to start updating correctly again

hartytp, 2019-09-04:
Yes, the effect is strongly dependent on the length of cable at Booster's output so most likely interference from the double reflection.

wizath, 2019-09-10:
Hmm, channel measurements are taken when channel power is enabled. That's the only limiting factor which can stop measurement task from running on selected channel. But again, if channel is interlocked, only RF switch is disconnected, channel has still enabled power

hartytp, 2019-09-10:
I'll write some code to fuzz Booster soon. That should help to identify any remaining issues like this.

> But again, if channel is interlocked, only RF switch is disconnected, channel has still enabled power

NB this even happened after the interlock was disconnected (see top post). Not sure if Booster stoped updating the power, or if the updates didn't make it to the VCP/SCPI interface. But, as I said, there is definitely a bug here somewhere; my power meter told me the power had changed, but the VCP interface lied and said it was the same. Asking the SCPI interface gave me a lie the first time, after which it started telling the truth...

wizath, 2019-09-10:
ok, I'll check it

hartytp, 2019-09-11:
any thoughts about this one? Otherwise I'll see if I can reproduce it with a HITL test (will take a day or two)

hartytp, 2019-09-13:
@wizath did you think of any way that this could occur? Why does a scpi read affect the vcp issues? Could this be linked to i2c errors somehow? Does an scpi power query trigger an i2c transaction?

hartytp, 2019-09-13:
Why would the power measurements stop updating? And why did an scpi power query make them start again? I don’t understand this

wizath, 2019-09-13:
All I can think of is some task blocking the measurement task - that would stop variables from updating. I'll run task tracing next week to check

hartytp, 2019-09-14:
@wizath i get that a task hanging could block the measurement cycle.

The thing I don’t understand is why a power query over scpi would clear the blockage. That’s odd isn’t it?

hartytp, 2019-09-14:
What I didn’t check was whether other scpi commands also unblocked the measurement loop. So maybe it’s possible that the issue here is the Ethernet task hanging and blocking measurements 

But if that’s the case does it mean that the software interlocks etc were also blocked? If so that’s serious!

Isn’t there a watchdog that reboots the fw if a task hangs?

hartytp, 2019-09-16:
@wizath so...this is the last known issue then (!!!) I still can't understand it. How could an SCPI command cause a task to unfreeze??

wizath, 2019-09-16:
I'm beginning my search for the cause of the problem

hartytp, 2019-09-17:
In order to help understand this, do you think we should add more diagnostics? If so, what can we do? Maybe a command that dumps the status of all tasks (including when they last updated) so we can see what has crashed/frozen/dead-locked/etc

hartytp, 2019-09-17:
hmm...looking over the code again, I really don't see how querying the output power could do anything, since that command doesn't interact with the hardware. Odd

hartytp, 2019-10-01:
@wizath did you have any more thoughts about what could have caused this, or any other diagnostics we can add to try to track it down if it reoccurs?

wizath, 2019-10-01:
I'll try to track this once I've complete v.1.4's for you

hartytp, 2019-10-01:
cool, thanks :)

wizath, 2019-10-01:
Anyway, ATM no idea what's going on.

hartytp, 2019-10-01:
Well, the firmware version I saw this on was quite old, so it's possible it was a symptom of some other bug. I'd suggest that you have a look at the code and see if you can find any source of the issue, or think of any more diagnostics we can implement. If not, we can close and re-open if it occurs again.

For me though, the smoking gun is why the VCP interface only started reporting correctly after an SCPI query, and why the first SCPI query gave the wrong value, but after that it worked. It's really odd because as far as I can tell the SCPI query doesn't interact with the hw directly, so should have no impact. Sounds like a deadlock between the threads or something.

wizath, 2019-10-01:
Maybe that was related to that weird bug with double to string conversion that caused HardFaults

wizath, 2019-10-01:
Anyway need to confirm that this still occurs in recent firmware

hartytp, 2019-10-01:
maybe. Anyway, please have a think about potential causes before we close. If there are any other diagnostics we can add (e.g. to look at last update time of each thread to check for deadlocks) then it might be worth doing that -- otherwise if I do reproduce it, we still won't have any new info...

hartytp, 2019-10-08:
@wizath did you have any thoughts about potential causes of this, or any extra diagnostics we should add?

hartytp, 2019-10-09:
@wizath I can't reproduce this. So, unless you can think of any potential cause or any additional diagnostics to add, please can you close?

wizath, 2019-10-09:
I'll check task runtime statistics and close

hartytp, 2019-10-09:
Cool, thanks. A VCP command to print task statistics would be awesome. Maybe we could even dump that on watchdog resets to see what task hung?

hartytp, 2019-10-09:
Maybe. Note that from my description of the issue above:
- I had the VCP in start mode
- I had an SCPI connection open, but no traffic
- I was getting data from the VCP, but the data wasn't updating (e.g. I could change the output power but always saw the same power and current reported)
- When I ran a few SCPI commands I started getting fresh data

So it seems that some task had stalled. So statistics on when each task last started/stopped would be really useful. If that were available after a watchdog it might also help to debug that (since we could see which task stalled)

hartytp, 2020-02-20:
Did the scpi log show anything?

hartytp, 2020-02-27:
I haven't seen this in ages and the FW has changed quite a bit since then. I think we will end up with a major firmware refactoring at some point in any case, so I don't think it's worth keeping this open any longer.

closed 2020-02-27 by hartytp