sinara-hw/Booster #271: bias current accuracy

hartytp, 2019-09-05:
Not an issue, but just curiosity... @wizath there is about 5mA of variation in the reported bias currents for our Booster v1.4. This is pretty low and not likely to be indicative of any issue, but I'm just curious where the variation comes from. Did they all read 50.0mA after calibration? Were sockets permuted after calibration? Do you think this is just temperature or something like that?

[image]

hartytp, 2019-09-05:
NB Booster reports 45mA, while using a high-resolution DMM I measure 4.473mV across the 0.1Ohm sense resistor So, that is a very good level of agreement.

Looks like the new InAmps work well.

hartytp, 2019-09-05:
(only open as I'm curious to hear @wizath's thoughts about the bias current calibration, not because there is an actual problem here)

wizath, 2019-09-05:
* As for my observations, that 5 mA current deviation is not really an issue.
* Current calibration usually ends up with 50-52 mA of channel current without RF switch enabled, based on lowest step size of current calibration algorithm (its like 10/4095)
* There is a temperature drift and we do not compensate for it in any means
* I think that everyone is instructed to do not swap channels between sockets

hartytp, 2019-09-05:
Okay, in that case, maybe this is just the resolution of the calibration. That's fine then.

Next time you calibrate a booster can you post the VCP log (`start`) here just so I know what it looks like immediately post calibration. Again, I don't suspect any issues, I'm just curious.

hartytp, 2019-10-16:
see https://github.com/sinara-hw/Booster/issues/315

closed 2019-10-16 by hartytp