sinara-hw/Booster #272: Booster should not start in standby mode

hartytp, 2019-09-05:
The v1.4 units we have still power on to stand-by mode, but I thought we'd agreed that they should boot into normal operation. Maybe I'm using an old version of the fw?

wizath, 2019-09-05:
Oh, I'm not yet done with VCP wiki section.
Use `powercfg 1` to set auto enable on startup. Thats configurable now.

hartytp, 2019-09-05:
Oh, nice! That's a good idea, thanks!

hartytp, 2019-09-05:
Can we make it so `powercfg 1` is the factory default setting though?

wizath, 2019-09-05:
I've changed it to 0 to prevent enabling all 8 channels in unknown state right after manufacturing. But if you really insist on switching it to 1 it's ok.

hartytp, 2019-09-05:
OK, that's fair enough. You're right, this is a sensible factory default behaviour.

So long as it's documented, it's easy for us to change this when we receive the devices so I'm happy to keep it as is.

closed 2019-09-05 by hartytp

wizath, 2019-09-05:
I'll put a note on wiki