sinara-hw/Booster #274: tx power -nan

hartytp, 2019-09-05:
[image]

not sure what I did to cause this

hartytp, 2019-09-05:
Sorry, misread the row headers, that's actually input power. This is okay, this is just clipping low.

Can you make this clip in a way that's consistent with the other channels? And make sure there is a note somewhere in the wiki about clipping? Thanks!

wizath, 2019-09-05:
You calibrated input power?
`inpcal <ch_num> 1 <input_power>` for 1st point
`inpcal <ch_num> 2 <input_power>` for 2nd point
I've used -5 and -20 dBm. I'll make that value to clip below -20dBm

hartytp, 2019-09-05:
No, I hadn't. Is this the error one gets if the input power is uncalibrated? If so, that's fine.

hartytp, 2019-09-05:
Sorry, forgot I had to do this :)

hartytp, 2019-09-06:
Leaving this open as a reminder to me that I should calibrate the input detector and give it a quick check. As far as I can tell, there is no actual bug or any other issue here, just me being confused :)

hartytp, 2019-10-09:
Tested as part of my fuzzing. Seems fine

closed 2019-10-09 by hartytp