sinara-hw/Booster #275: corrupted serial port

hartytp, 2019-09-05:
at least once, the USB connection on my serial port came loose while I was in debug mode (had send the `start` command) and I started getting corrupt data back:
[image]

hartytp, 2019-09-05:
hmm...can't reproduce this now however so not sure exactly what happened

wizath, 2019-09-05:
Weird one, minicom always shows `cannot open /dev/ttyACM1` when accidental unplug happens

hartytp, 2019-09-05:
Sorry, yes. It showed that, then plugged back in and reopened the terminal and got strange behaviour

wizath, 2019-09-07:
This seems more like terminal emulator error than Booster issue. Never happened to me 🤔

If you reopened that means you closed and reopened the terminal app? Which one do you use?

hartytp, 2019-09-07:
Mini term completely closed miniterm and reopened. Seemed like some buffering issue on the booster vcp. It also stopped echoing properly (felt like it was printing characters from a few before the one I typed as if the data it was sending wasn’t properly synchronised)

hartytp, 2019-09-07:
I couldn’t reproduce this so unless you can think of a reason it might have happened I suggest we close for now and reopen if it occurs again.

wizath, 2019-09-07:
Not really. With most recent TS chasis there is nice cutout for USB so it sits tight.

hartytp, 2019-09-07:
Okay, well it only affects the vcp with isn’t the main tnterface and I can’t reproduce so closing for now

Will reopen if it happens again

closed 2019-09-07 by hartytp