sinara-hw/Booster #277: calibration question

hartytp, 2019-09-06:
The wiki says
```
`[biascal] error, current too high 52 > 50`

This is related to module current exceeding 2% setpoint margin because DAC step is set a little bit high. It'll be fixed soon, but for now, you can just redo your calibration with slightly higher current eg. 52 mA and it should be fine

```

Is this still an issue?

wizath, 2019-09-07:
Oops, yes. Not really an issue tho, but fix will clearly improve user experience.

wizath, 2019-09-10:
Lowered the step size when value is near desired current, now works fine

closed 2019-09-10 by wizath

hartytp, 2019-09-10:
Great.

Do you think you can finish the wiki docs today? I'm very keen to finish testing...

wizath, 2019-09-10:
Wiki is pretty much finished, unless you have any requests. I've updated calibration section yesterday and today I'll just add examples of VCP commands

hartytp, 2019-09-10:
Great!. Once you think it's completely finished let me know and I'll read.